Entware/Entware #1: arm-openwrt-linux-gnueabi-gcc-ar binary not included in gcc package

iacchi, 2018-03-09:
### Package: gcc

- expected behavior: arm-openwrt-linux-gnueabi-gcc-ar binary present in the package
- actual behavior: binary is missing, compiling of program fails

The gcc package in Entware doesn't come with the binary wrappers for ar and nm. Some sources require these binaries to be present in order to compile properly. For example, installing the numpy python module with pip fails with the following error: "/bin/ash: arm-openwrt-linux-gnueabi-gcc-ar: command not found" Have a look here to see what these binary wrappers are: https://stackoverflow.com/questions/48777554/what-is-the-difference-between-ar-nm-and-gcc-ar-gcc-nm
I am wondering if it's possible to provide these wrappers in the package you provide. Looking at the content of your gcc package, inside folder /opt/lib/gcc/arm-openwrt-linux-gnueabi/6.3.0/ there is a binary file called lto-wrapper that might do what is needed, but I've tried to link it on /opt/bin with the name arm-openwrt-linux-gnueabi-gcc-ar and compilation of numpy is still failing with error "arm-openwrt-linux-gnueabi-gcc-ar: fatal error: environment variable COLLECT_GCC must be set".
Alternatively, since it's only a wrapper, I am wondering if it's possible to create a shell script that will behave as needed (again, look at the stackoverflow thread).

### Platform: armv7

Device: Synology DS214+

zyxmon, 2018-03-09:
1.Is ar package installed? (`opkg list-installed` ) 
2. Try to patch your makefile to use ar instead of  arm-openwrt-linux-gnueabi-gcc-ar or make a symlink
`ln -s /opt/bin/ar /opt/bin/arm-openwrt-linux-gnueabi-gcc-ar`.

iacchi, 2018-03-09:
ar is installed. I am trying to install a package from pip, so I don't know how to patch it exactly. Regardless, update would have the same issue, so I think it's best to fix this in our end. A simple symlink wouldn't work because there would be a missing parameter. I will try to figure things out and in case I will report back here.

zyxmon, 2018-03-09:
Try this one - http://bin.entware.net/armv7sf-k3.2/test/arm-openwrt-linux-gnueabi-gcc-ar

iacchi, 2018-03-09:
worked like a charm, thanks :) You may want to think about packaging it in the gcc package.

zyxmon, 2018-03-12:
packaging changed - https://github.com/Entware/entware-packages/commit/bf84e499dcf7a12b51f08afdf0577a8d90df59a7

closed 2018-03-12 by zyxmon