Entware/Entware #3: ip utility error with running command in namespace

LeftPinkie, 2018-03-11:
### Package: ip-full

For the `ip` utility in `ip-full` package, I am having issues running commands in namespace.

- expected behavior:
```
# Create the network namespace
ip netns add test_ns

# Create the virtual network pair
ip link add test_v0 type veth peer name test_v1

# Move test_v0 to the test_ns namespace
ip link set test_v0 netns test_ns

# Configure the ends of the virtual network pairs
ip netns exec test_ns ifconfig test_v0  10.1.1.10/24 up
```
- actual behavior:
I get an error running `ip netns exec test_ns ifconfig test_v0  10.1.1.10/24 up`. The error message is:
setting the network namespace "test_ns" failed: Function not implemented

I cannot even run a basic command:
`ip netns exec test_ns ip addr list`

### Platform:
Linux 4.2.8
ip utility, iproute2-ss4.4.0-10-entware

zyxmon, 2018-03-11:
Please check if your linux kernel is compiled with namespaces support.

```
CONFIG_NAMESPACES=y
...
CONFIG_NET_NS=y
```

LeftPinkie, 2018-03-11:
I cannot find the `.config` file.  This is a NAS from QNAP so I do not know how the kernel was compiled.  Any idea as to how I can query for this info?

zyxmon, 2018-03-11:
Just google for the answer. iproute2 error indicates there is no namespace support in the kernel.

closed 2018-03-11 by LeftPinkie