Entware/Entware #4: MiniDLNA, Dovecot: file not found /opt/etc/passwd

blackfuel, 2018-03-14:
### Package: MiniDLNA and Dovecot
Since the new Entware merge, I now must do this once before starting MiniDLNA or Dovecot.  These programs try to lookup the user in this password file `/opt/etc/passwd`, and it fails to start.  My workaround fix links to the real one.
`for F in passwd group shadow gshadow; do ln -sf /etc/$F /opt/etc/$F; done`

### Platform: RT-AC68U Asuswrt-Merlin
Entware armv7-2.6 compiled from source.

zyxmon, 2018-03-14:
How have you installed Entware? These simlinks should be created by install script?

The only scenario when these symlinks are missing is armv7 is probably when you have used asuswrt-install.sh script for armv7 that was given in snbforum topic.
In this case I'd recomend to run
[code]
opkg remove asuswrt-opt
opkg install entware-opt
wget http://bin.entware.net/other/symlinks.sh -O- | sh
[/code]
Some more symlinks will be created. Your installation will be correct.

blackfuel, 2018-03-15:
Running the `symlinks.sh` script fixed my problem.  I use a local Entware repo that I compiled myself.  Just recently I switched from Entware-ng to the new Entware.

closed 2018-03-15 by blackfuel