Entware/Entware #5: Support for MIPS, not just MIPSEL

ian5142, 2018-03-14:
Is there planned support for MIPS, not just MIPSEL. I am looking for MIPS support such as those found in Atheros based routers, TP-Link Archer C7 v2, etc.

zyxmon, 2018-03-14:
It is here - http://bin.entware.net/mipssf-k3.4/
Installers - http://bin.entware.net/mipssf-k3.4/installer/

zyxmon, 2018-03-14:
Entware-3x users will upgrade to Entware automatically running `opkg update; opkg upgrade` 2 times.
If there are problems with upgrade - please report!

ian5142, 2018-03-14:
Why is it not located in the same location as the other installers for MIPSEL and ARM? 
See the wiki page here on github: https://github.com/Entware/Entware/wiki/Install-on-DD-WRT

zyxmon, 2018-03-14:
Wiki pages are old. All new installers are on bin.entware.net.

ian5142, 2018-03-14:
I am maintaining a wiki page on the dd-wrt wiki here: https://www.dd-wrt.com/wiki/index.php/Installing_Entware-3X

zyxmon, 2018-03-14:
Thanks for DD-WRT wiki.
The project now is call Entware without any suffix. The complete list of installers is here
http://forums.zyxmon.org/viewtopic.php?p=17841#p17841

I hope we will update our wiki soon.

closed 2018-03-14 by zyxmon