Entware/Entware #6: Package: perl-DBD-mysql

ivailop, 2018-03-15:
### Package: perl-DBD-mysql

For new package to be added:

To be able to host [Bugzilla](https://www.bugzilla.org/) on our QNAP server we need [DBD-mysql](http://search.cpan.org/~capttofu/DBD-mysql/).

It is not possible to build locally as it requires `libmysqlclient-dev` which could not be found anywhere.

### Platform:

QNAP TS-253 Pro, version 4.34.0486

zyxmon, 2018-03-15:
> It is not possible to build locally as it requires libmysqlclient-dev which could not be found anywhere.
There are no *-dev packages in entware, but it is possible to build a lot of perl & python modules. Wiki is  will be updated soon. But it is mostly correct.
You must install gcc, headers and mariadb package to build this perl module.

ivailop, 2018-03-15:
Yes, I build many other Perl modules with no problem but that one I could not due to missing `dev` (include  and .a) files.

zyxmon, 2018-03-16:
You do not need .a file.
Headers for x64 are here - http://bin.entware.net/x64-k3.2/include/include.tar.gz
Old wiki - https://github.com/Entware/Entware/wiki/Self-installation-of-perl-modules
Please use the headers above. Do not use headers from wiki (native gcc section).

zyxmon, 2018-03-23:
Wiki is updated. I have successfully installed this prel module using wiki

```
.......
All tests successful.
Files=73, Tests=9, 23 wallclock secs ( 0.48 usr  0.16 sys + 20.35 cusr  1.75 csys = 22.74 CPU)
Result: PASS
  CAPTTOFU/DBD-mysql-4.046.tar.gz
  /opt/bin/make test -- OK
Running make install
"/share/CACHEDEV1_DATA/.qpkg/Entware/bin/perl5.26.1" -MExtUtils::Command::MM -e 'cp_nonempty' -- mysql.bs blib/arch/auto/DBD/mysql/mysql.bs 644
Files found in blib/arch: installing files in blib/lib into architecture dependent library tree
Installing /opt/lib/perl5/5.26/auto/DBD/mysql/mysql.so
Installing /opt/lib/perl5/5.26/DBD/mysql.pm
Installing /opt/lib/perl5/5.26/DBD/mysql/INSTALL.pod
Installing /opt/lib/perl5/5.26/DBD/mysql/GetInfo.pm
Installing /opt/lib/perl5/5.26/Bundle/DBD/mysql.pm
Appending installation info to /opt/lib/perl5/5.26/perllocal.pod
  CAPTTOFU/DBD-mysql-4.046.tar.gz
  /opt/bin/make install  -- OK

```

closed 2018-03-23 by zyxmon