Entware/Entware #10: Package request: ewf-tools

skysbsb, 2018-03-16:
### Package: ewf-tools

For existing package:

- expected behavior: needed package to work with Expert Witness Compression Format (EWF) and make E01 images of USB external devices in my QNAP storage.

- make sure there's no similar package already added, 
ok

- describe killer features of new package,
Read or write supported EWF formats:
SMART .s01 (EWF-S01)
EnCase .E01 (EWF-E01) and .Ex01 (EWF2-Ex01)
Read-only supported EWF formats:
Logical Evidence File (LEF) .L01 (EWF-L01) and .Lx01 (EWF2-Lx01)
The libewf package contains the following tools:
ewfacquire, which writes storage media data from devices and files to EWF files.
ewfacquirestream, which writes data from stdin to EWF files.
ewfdebug; experimental tool does nothing at the moment.
ewfexport, which exports storage media data in EWF files to (split) RAW format or a specific version of EWF files.
ewfinfo, which shows the metadata in EWF files.
ewfmount, which FUSE mounts EWF files.
ewfrecover; special variant of ewfexport to create a new set of EWF files from a corrupt set.
ewfverify, which verifies the storage media data in EWF files.
ewfaquirestream-mt, C++ 11 multi-threaded version of ewfacquirestream contributed by Bernhard Zach in 2013.

- make sure it's open source and not abandoned (at least for last 5 years or so),
https://github.com/libyal/libewf/ - last update a month ago

- please, use native compilation for additional PHP, Perl, Ruby, Python modules for non-MIPS platforms.

### Platform:

- please, tell us what feed and\or devices are you using.
QNAP STORAGE, model TVS-1271U-RP, x86_64

- be ready to check issue-related  changes or it will be closed after 30 days with "no response" status.
ok

skysbsb, 2018-03-17:
Hey, I generated ewf-tools binary for x86_64 using a merge/mess of some packages Makefiles presented at Entware package folder.. I am sure this Makefile is not right at all, but solved my problem (generated ewf tolls binary). But this can be the start of something that can end by someone else.
[Makefile2.txt](https://github.com/Entware/Entware/files/1822020/Makefile2.txt)

ryzhovau, 2018-09-13:
Are you sure no any additional kernel modules required?

closed 2018-09-27 by no-response[bot]