Entware/Entware #11: Package: OLSR

colindean, 2018-03-16:
_(carrying over from https://github.com/Entware/Entware-ng/issues/696)_

I would like to see OLSR and some other mesh routing systems included in Entware. I would like this for @pittmesh / @metamesh and other meshnet projects to use.

I searched around and cannot seem to find where the binaries produced from https://github.com/Entware/entware-routing are hiding.

I see that entware-routing is enabled in feeds.conf: https://github.com/Entware/Entware/blob/master/feeds.conf#L2

It looks like `olsrd` may just need to be enabled in the configs for each architecture. It looks like it's just commented out: 

https://github.com/Entware/Entware/blob/9b47f669e875d462e7b37ed1f98990f129c78d17/configs/armv7-3.2.config#L4366

closed 2019-04-26 by ryzhovau

colindean, 2019-04-27:
Thank you, @ryzhovau!

ryzhovau, 2019-04-27:
This issue was opened year ago, sorry for keep you waiting.

Just checked binaries are working. I'll be glad to hear a feedback from you: config fixes, your own start script or any other info to improve package.