Entware/Entware #7: Package Request: cachefilesd

KyleSanderson, 2018-03-16:
Package: cachefilesd
make sure there's no similar package already added,
Done.

describe killer features of new package,
Provides caching for already mounted filesystems. This is an ancient package that has been recommended by RedHat for years to improve NFS locality performance.

make sure it's open source and not abandoned (at least for last 5 years or so),
This package is actively maintained by David Howells (dhowells) on kernel.org
https://git.kernel.org/pub/scm/linux/kernel/git/dhowells/cachefilesd.git/

Platform:
please, tell us what feed and\or devices are you using.
root@Tomato:/tmp/home/root# cat /proc/cpuinfo
Processor : ARMv7 Processor rev 0 (v7l)
processor : 0
BogoMIPS : 1998.84
processor : 1
BogoMIPS : 1998.84

Features : swp half thumb fastmult edsp
CPU implementer : 0x41
CPU architecture: 7
CPU variant : 0x3
CPU part : 0xc09
CPU revision : 0

Hardware : Northstar Prototype
Revision : 0000
Serial : 0000000000000000

root@Tomato:/tmp/home/root# opkg update
Downloading http://pkg.entware.net/binaries/armv7/Packages.gz.
Updated list of available packages in /opt/var/opkg-lists/packages.
root@Tomato:/tmp/home/root#

be ready to check issue-related changes or it will be closed after 30 days with "no response" status.
Sure.

KyleSanderson, 2018-03-16:
Looks like @ryzhovau did a drive-by on the old repo, re-created the issue here as per https://github.com/Entware/Entware-ng/issues/640

zyxmon, 2018-09-13:
@KyleSanderson - I hope you have already upgraded Entware-ng to Entware.
Here is a package to be tested - http://bin.entware.net/armv7sf-k2.6/test/cachefilesd_0.10.10-1_armv7-2.6.ipk

closed 2018-09-28 by no-response[bot]