Entware/Entware #15: Package request : radicale

jemail99, 2018-03-17:
### Package: radicale

http://radicale.org/

Radicale is a small but powerful CalDAV (calendars, todo-lists) and CardDAV (contacts) server, that:

    Shares calendars through CalDAV, WebDAV and HTTP.
    Shares contacts through CardDAV, WebDAV and HTTP.
    Supports events, todos, journal entries and business cards.
    Works out-of-the-box, no installation nor configuration required.
    Can warn users on concurrent editing.
    Can limit access by authentication.
    Can secure connections.
    Works with many CalDAV and CardDAV clients.
    Is GPLv3-licensed free software. 

### Platform:
Device Asus RT-AC68U running merlin firmware.

zyxmon, 2018-03-20:
You can easily install it yourself. See wiki -https://github.com/Entware/Entware/wiki/Self-installation-of-python-modules

```
opkg install python3-pip
pip3 install --upgrade pip setuptools
python3 -m radicale --help
usage: radicale [OPTIONS]

optional arguments:
  -h, --help            show this help message and exit
  --version             show program's version number and exit
  --verify-storage      check the storage for errors and exit
  -C CONFIG, --config CONFIG
                        use a specific configuration file

server:
  -H SERVER_HOSTS, --hosts SERVER_HOSTS, --server-hosts SERVER_HOSTS
                        set server hostnames including ports
.....

```

closed 2018-03-20 by zyxmon

jemail99, 2018-04-18:
Sorry to re-open this, but installing it is not the problem.
The problem is getting bcrypt htpassword hashing working and compiling to work with radicale. Maybe someone more knowledgeable than myself can check if it's possible to set up radicale on a router with bcrypt encryption module functioning so that secure logins can be made over the internet to the radicale server?

RogueScholar, 2018-09-19:
@jemail99 I know this comes rather belatedly but I just saw this issue. A glance at the documentation seems to suggest that to enable that functionality you would first need to issue these commands:
`opkg install libcryptoxx
pip3 install --upgrade passlib bcrypt --upgrade-strategy eager`
If that package and those two modules install successfully then a simple reboot of the router should be all that remains for you to be able to start Radicale and have it operate the way you wish.