Entware/Entware #17: Package request: gkrellmd

DrCaFr, 2018-03-17:
## Package gkrellmd:

I am using gkrellm on both, Linux and Windows since many years to monitor performance of local and remote systems.
"Killer features": gkrellm is the most versatile and cross-platform, but ressource efficient monitor tool I know.
The corresponding daemon (has no GUI) is called gkrellmd.
For details, see http://gkrellm.srcbox.net/.
Today, I upgraded my QNAP NAS from QTS 4.2 to 4.3.
QTS 4.3 does not longer support IPKG (which contained gkrellmd) and 32 bit.
Recommended IPKG-replacement Entware-ng does not yet contain gkrellmd.
Since I desperately miss it, please add gkrellmd!

## Platform:

I am using QNAP TS-870 with x64 CPU

pocketgeek, 2018-03-20:
I second this.  I'd really like to be able to load this on my arm based Asus routers.

zyxmon, 2018-03-20:
Package to test
x64 - http://bin.entware.net/x64-k3.2/test/gkrellmd_2.3.10-1_x64-3.2.ipk
arm (asus, 2.6.36) - http://bin.entware.net/armv7sf-k2.6/test/gkrellmd_2.3.10-1_armv7-2.6.ipk

Please check if it works or not.
Please create startup script (entware style) if possible, If conf file needs modification - please tell us.

pocketgeek, 2018-03-21:
Works well for me.  Conf file is good.  I'll have a go at creating a proper init script for it.

zyxmon, 2018-03-21:
Thanks for testing. This startup scrip looks like correct - https://github.com/Entware/entware-oldpackages-ports/blob/master/gkrellmd/files/S30gkrellmd

pocketgeek, 2018-03-21:
That init script seems to work perfect.

Thanks!

zyxmon, 2018-03-21:
gkrellmd is in the repo - https://github.com/Entware/entware-oldpackages-ports/commit/07df6496e000463da380d897e48171d6582f8b54

closed 2018-03-21 by zyxmon

DrCaFr, 2018-03-21:
Hi zyxmon,

thanks a lot for the good news! I received it while travelling and was quite eager to try the new software.
Unfortunately, I cannot install <http://bin.entware.net/x64-k3.2/test/gkrellmd_2.3.10-1_x64-3.2.ipk> on my QNAP. Error message reads:

> opkg install gkrellmd_2.3.10-1_x64-3.2.ipk
> Unknown package 'gkrellmd'.
> Collected errors:
>  * pkg_hash_fetch_best_installation_candidate: Packages for gkrellmd found, but incompatible with the architectures configured
>  * opkg_install_cmd: Cannot install package gkrellmd.

I believe, x64 is the correct architecture, since:
> opkg  print-architecture
> arch all 1
> arch noarch 1
> arch x86-64 10

What can I do, now?

Thanks a lot in advance, 

  Carsten

zyxmon, 2018-03-22:
You are still using Entware-3x. Please upgrade to Entware - https://forum.qnap.com/viewtopic.php?f=351&t=139781

DrCaFr, 2018-03-22:
Great! This **really** works great!
Thanks a lot!
Большое спасибо!