Entware/Entware #18: Package request: transmission (update)

mariozski, 2018-03-17:
### Package: transmission

Current transmission version (2.92) has a quite significant security vulnerability (https://github.com/transmission/transmission/pull/468) that has been been fixed in new (2.93) version.

### Platform:

I am using Asus RT-AC56U with Asuswrt-Merlin.

zyxmon, 2018-03-17:
Current transmission in Entware is 2.93. My guess - you are still using Entware-ng.

closed 2018-03-17 by zyxmon

itgold, 2019-02-04:
I can still see transmission is version 2.92+git-5
Is that the right URL to look at: https://pkg.entware.net/binaries/armv7/Packages.html ?

Any way to upgrade it to 2.93 manually?

zyxmon, 2019-02-04:
@itgold - you are using entware-ng that is not supported and updated.
Please use Entware if you hardware supports Entware.