Entware/Entware #22: Alternative installation rewrites /etc/passwd

ValdikSS, 2018-03-18:
### Package: entware-upgrade

armv7 (kernel >= 2.6.36) alternative installation script rewrites /etc/passwd, most probably because of code in entware-upgrade postinst script.
It seems that /opt/etc/passwd already exist when entware-upgrade postinst is executed, that's why it relinks to /etc/passwd, and later becomes rewritten from the installation script.

Code from entware-upgrade:
```
# now try create symlinks - it is a std installation
if [ -f /etc/passwd ]
then
    ln -sf /etc/passwd /opt/etc/passwd
fi
```

Code from installation script:
```
# now copy default files - it is an alternative installation
cp -f /opt/etc/passwd.1 /opt/etc/passwd
```

- expected behavior: /opt/etc/passwd is not symlinked to /etc/passwd and stores independent records.
- actual behavior: /etc/passwd gets rewritten by /opt/etc/passwd.1 contents.

ValdikSS, 2018-03-18:
Simple workaround is to remove /opt/etc/passwd after `opkg install entware-opt` in installation script.

zyxmon, 2018-03-18:
Please read the wiki - https://github.com/Entware/Entware/wiki/Alternative-install-vs-standard

ValdikSS, 2018-03-18:
@zyxmon I'm talking about *alternative* installation, which currently rewrites original /etc/passwd (not /opt/etc/passwd) due to bug. It should not do that.

zyxmon, 2018-03-18:
>> It should not do that.

Sure - what script you are talking about?

zyxmon, 2018-03-18:
Yes - I'll check it.

ValdikSS, 2018-03-18:
http://bin.entware.net/armv7sf-k2.6/installer/alternative.sh, from the wiki. After `opkg install entware-opt` /opt/etc/passwd is symlinked to /etc/passwd, and later on the data from /opt/etc/passwd.1 gets copied to /opt/etc/passwd, and eventually rewrites /etc/passwd.

zyxmon, 2018-03-18:
Yes I see - script used to upgrade Entware-ng (which is always standard) is executed for alternative armv7-k2.6.
I'll fix it. Thanks.  It looks like root/12345 (/opt/etc/passwd) is rewritten, /etc/passwd is not affected.

Завтра исправлю на свежую голову. Пока считаю, что ошибка есть, но описана неверно.

ValdikSS, 2018-03-18:
Скрипт ведет себя именно так, как я написал в первом сообщении — альтернативный инстраллятор переписывает /etc/passwd во время установки. Перед выполнением кода `cp -f /opt/etc/passwd.1 /opt/etc/passwd` из инсталлятора, /opt/etc/passwd является симлинком на /etc/passwd, и выполнение `cp` перезаписывает данные /etc/passwd.

zyxmon, 2018-03-18:
Выполнение cp переписывает /opt/etc/passwd, после cp /opt/etc/passwd уже не симлинк, а нормальный файл, причем тот, что нужно.
У Вас что за устройство и что за прошивка?

ValdikSS, 2018-03-18:
Команда cp всегда следует симлинку, и перезаписывает файл, на который ссылается симлинк. Я только что проверил это и на Fedora 27, и на busybox.
У меня нестандартное Linux-окружение, собранное с Android NDK и Bionic, и с Android init-системой.

zyxmon, 2018-03-18:
Поведение команды cp в openwrt и tomato другое. Все зависит от флага по умолчанию
-P - preserve symlinks;
-H - follow symlinks.
http://man7.org/linux/man-pages/man1/cp.1.html

Поэтому при тестировании на Tomato проблема не обнаружилась. А она есть.

zyxmon, 2018-03-18:
http://bin.entware.net/armv7sf-k2.6/installer/alternative.sh - was fixed.
This is specific bug only for this arch, that comes from entware-ng.

closed 2018-03-18 by zyxmon

zyxmon, 2018-03-19:
BTW (busybox sources).
```
/* coreutils 6.9 compat:
         * by default, "cp" derefs symlinks (creates regular dest files),
         * but "cp -R" does not. We switch off deref if -r or -R (see above).
         * However, "cp -RL" must still deref symlinks: */

```
Thanks again. Installation script now removes symlinks, that were created for standard install by one of the packages before copying regular files.

ryzhovau, 2018-03-23:
@ValdikSS glad you are using Entware:)

ValdikSS, 2018-03-25:
Entware is now available on Huawei portable LTE routers with modified firmware.
E5372: https://4pda.ru/forum/index.php?s=&showtopic=618520&view=findpost&p=64916777
E5770: https://4pda.ru/forum/index.php?s=&showtopic=818849&view=findpost&p=64185933

It uses alternative installation. Entware busybox ash is available with 'entware shell' command to prevent PATH and LD_LIBRARY_PATH mixing.
I had to `touch /opt/usr/lib/locale/locale-archive` before installation to prevent locale-gen from running, since it will be killed by oom-killer *with the installation script*, since the device have only 15-16 MB of free RAM. I download binary locale-archive later, after the installation.