Entware/Entware #25: jdupes

51DD5, 2018-03-19:
### Package: jdupes

For new package to be added:

Please add jdupes. Many users seem to be relying on fdupes for file de-duplication. By the sound of it fdupes development has ground to a halt and the baton has bee picked up by Jody Bruchon with the forked project jdupes: https://github.com/jbruchon/jdupes, which is also much faster, more efficient and more powerful than fdupes and would prove a useful tool. I personally would like to see it added to carry out de-duplication on my QNAP TS-451 NAS, where fdupes took a long time, then crashed at 50%!
Thanks ;-).

ryzhovau, 2018-09-13:
Should we remove `fdupes` from repo after `jdupes` will be available?

jrgutier, 2018-09-16:
> Should we remove `fdupes` from repo after `jdupes` will be available?

I believe you should.

51DD5, 2018-10-01:
Hi, sorry, I opened this package request so long ago I took my eye off the ball and missed your question @ryzhovau . Look, I'm afaraid I don't know if fdupes is still popular and gets used as a package, but it would seem to make sense to remove it if it is no longer supported and has been improved upon with jdupes.

closed 2018-10-02 by ryzhovau