Entware/Entware #27: Package request: Chromaprint

gravelld, 2018-03-19:
### Package: Chromaprint
Chromaprint is the open source client-side library for the Acoustid project. It implements a custom algorithm for extracting fingerprints from any audio source.

Chromaprint is used as the fingerprinting library by a number of software tools:

- beets
- bliss
- filebot
(Those off the top of my head).

Chromaprint has already been built and is used on Synology devices which use similar architectures. [It has also been added to Optware-ng](https://github.com/Optware/Optware-ng/commit/e2d6436fa0b7c22c9410c636cd9e2c9e70b8d3f4).

### Platform:
Most important are:

- ARMv7
- x86_64

ryzhovau, 2018-09-13:
We didn't keep libraries with no dependent application. Which application you will use?
AFAIK, beets is just a python script without chromaprint dependency, [tried](https://asciinema.org/a/928) 6 years ago.

gravelld, 2018-09-13:
> We didn't keep libraries with no dependent application
I don't quite understand what you mean, sorry. Do you mean you don't add libraries that don't have an application that _absolutely requires_ it?

beets doesn't _require_ Chromaprint, and neither do the other tools listed, but they can use it.

ryzhovau, 2018-09-13:
> I don't quite understand what you mean, sorry. Do you mean you don't add libraries that don't have an application that absolutely requires it?

Yes, we don't keep libraries which is not required by some other package(s) from repo. 

> beets doesn't require Chromaprint, and neither do the other tools listed, but they can use it.

I suspect `chromaprint` contains some additional binaries, not a library only, because `bliss` and `filebot` written in Java and can't be linked to C library directly. If so, I can add `chromaprint` to Entware.

ryzhovau, 2018-09-13:
Yep, there's an utility to generate fingerprints form audio files. I'll add `chromaprint` as standalone package.

gravelld, 2018-09-13:
That's right - `fpcalc` can be used to generate the fingerprints. I suppose I was a bit loose with my wording of "library".

ryzhovau, 2018-09-13:
Quite heavy for MIPSEL devices:

* ~23 seconds to generate fingerprint on single 10MB MP3 file,
* ~22 seconds to generate fingerprint on single 30MB FLAC file.

ryzhovau, 2018-09-13:
Binaries will arrive soon.

closed 2018-09-13 by ryzhovau