Entware/Entware #28: Package Request: libfuse-dev

ihenry, 2018-03-20:
### Package: libfuse-dev

For existing package: libfuse

Trying to install rar2fs
https://hasse69.github.io/rar2fs/

But it requires the libfuse-dev package.
I found the libfuse package, but to compile I need the dev package.

### Platform:
QNAP x64

zyxmon, 2018-03-20:
The are no *-dev packages in Entware with few exceptions. You need to download and unpack headers manually - https://github.com/Entware/Entware/wiki/Using-GCC-for-native-compilation
Please modify download URL to fit your arch.

closed 2018-03-20 by zyxmon

kaztost, 2018-03-20:
BTW, I hope the package rar2fs (http://bin.entware.net/x64-k3.2/rar2fs_1.26.0-1_x64-3.2.ipk) is working. Or no?

ihenry, 2018-03-20:
Many Thanks.
I started off looking for the package and I couldn't find it, so I then looked to compile.
You have saved me further pain with the required dependencies- Thanks again.
rar2fs is now installed. 

`# opkg install rar2fs
Installing rar2fs (1.26.0-1) to root...
Downloading http://pkg.entware.net/binaries/x86-64/rar2fs_1.26.0-1_x86-64.ipk
Installing libunrar (5.5.8-1) to root...
Downloading http://pkg.entware.net/binaries/x86-64/libunrar_5.5.8-1_x86-64.ipk
Configuring libunrar.
Configuring rar2fs.
`