Entware/Entware #30: package: iw missing from bin.entware.net

dwfreed, 2018-03-21:
### Package: iw

iw is present in this repository, but is completely missing from bin.entware.net (checked other platforms too)

### Platform:

armv7

zyxmon, 2018-03-21:
Not all openwrt packages were added to Entware. This one looks hardware specific. We will try to add it if there will be more requests.

closed 2018-03-21 by zyxmon

dwfreed, 2018-03-22:
It just seems weird that it's in this repo, but ipks aren't built for it.  It's hardware specific in that it's only useful if you have a wireless interface of some kind, but I'd guess that a large majority of people that use Entware are doing so on their wireless routers.

Triaholgi, 2018-08-23:
Is it possible to reopen the issue? 
Why: With my Asus rt-ac87u I'm under merlin-fw (384.5-0) and have newest entware (and debian jessie in chrooted enviroment) running. My plan is, that I want to establish a mesh network w/ batman-adv (https://www.open-mesh.org/projects/batman-adv/wiki/Quick-start-guide). To do this 
- I need the correspondig kernel modules -> can compile it on my own (https://hqt.ro/how-to-compile-modules-for-asuswrt-routers) -> no issue
- I need to reconfigure wifi interfaces. At least I need the 'iw' command (ip cmd is there) -> issue

 I should note that under chrooted debian env it is possible to install 'iw' over apt. But I get follwing error:
`root@router:/# iw list
nl80211 not found.`

Is it possible to get 'iw' work under entware?

i0ntempest, 2020-05-04:
+1 for this package. I need this for airman-ng.

ryzhovau, 2020-05-04:
Please note, there's no any unified API used by `wi` in embedded world, most of vendors ignores nl80211netlink interface. This utilities may be useful just in few cases, mostly on x86 platforms.