Entware/Entware #33: MC packaging error

kaztost, 2018-03-22:
On aarch64 and armv7 everything were fine. But for mipsel:
```
kaztost@ubuntu:~/Entware$ make package/mc/compile V=s
make[1]: Entering directory `/home/kaztost/Entware'
make[2]: Entering directory `/home/kaztost/Entware/package/libs/ncurses'
make[2]: Leaving directory `/home/kaztost/Entware/package/libs/ncurses'
make[2]: Entering directory `/home/kaztost/Entware/package/libs/toolchain'
make[2]: Leaving directory `/home/kaztost/Entware/package/libs/toolchain'
make[2]: Entering directory `/home/kaztost/Entware/package/libs/zlib'
make[2]: Leaving directory `/home/kaztost/Entware/package/libs/zlib'
make[2]: Entering directory `/home/kaztost/Entware/package/libs/ncurses'
make[2]: Leaving directory `/home/kaztost/Entware/package/libs/ncurses'
make[2]: Entering directory `/home/kaztost/Entware/feeds/oldports/slang2'
make[2]: Leaving directory `/home/kaztost/Entware/feeds/oldports/slang2'
make[2]: Entering directory `/home/kaztost/Entware/feeds/packages/utils/attr'
make[2]: Leaving directory `/home/kaztost/Entware/feeds/packages/utils/attr'
make[2]: Entering directory `/home/kaztost/Entware/feeds/packages/libs/libffi'
make[2]: Leaving directory `/home/kaztost/Entware/feeds/packages/libs/libffi'
make[2]: Entering directory `/home/kaztost/Entware/package/libs/gettext-full'
make[2]: Leaving directory `/home/kaztost/Entware/package/libs/gettext-full'
make[2]: Entering directory `/home/kaztost/Entware/package/libs/libiconv'
make[2]: Leaving directory `/home/kaztost/Entware/package/libs/libiconv'
make[2]: Entering directory `/home/kaztost/Entware/feeds/packages/libs/glib2'
make[2]: Leaving directory `/home/kaztost/Entware/feeds/packages/libs/glib2'
make[2]: Entering directory `/home/kaztost/Entware/feeds/packages/libs/libffi'
make[2]: Leaving directory `/home/kaztost/Entware/feeds/packages/libs/libffi'
make[2]: Entering directory `/home/kaztost/Entware/package/libs/gettext'
make[2]: Leaving directory `/home/kaztost/Entware/package/libs/gettext'
make[2]: Entering directory `/home/kaztost/Entware/package/libs/gettext-full'
make[2]: Leaving directory `/home/kaztost/Entware/package/libs/gettext-full'
make[2]: Entering directory `/home/kaztost/Entware/package/libs/libiconv-full'
make[2]: Leaving directory `/home/kaztost/Entware/package/libs/libiconv-full'
make[2]: Entering directory `/home/kaztost/Entware/feeds/packages/libs/glib2'
make[2]: Leaving directory `/home/kaztost/Entware/feeds/packages/libs/glib2'
make[2]: Entering directory `/home/kaztost/Entware/package/utils/util-linux'
make[2]: Leaving directory `/home/kaztost/Entware/package/utils/util-linux'
make[2]: Entering directory `/home/kaztost/Entware/feeds/packages/utils/mc'
rm -rf /home/kaztost/Entware/build_dir/target-mipsel_mips32r2_glibc-2.27/mc-4.8.20/.pkgdir/mc.installed /home/kaztost/Entware/build_dir/target-mipsel_mips32r2_glibc-2.27/mc-4.8.20/.pkgdir/mc
mkdir -p /home/kaztost/Entware/build_dir/target-mipsel_mips32r2_glibc-2.27/mc-4.8.20/.pkgdir/mc
install -d -m0755 /home/kaztost/Entware/build_dir/target-mipsel_mips32r2_glibc-2.27/mc-4.8.20/.pkgdir/mc/opt/bin
install -m0755 /home/kaztost/Entware/build_dir/target-mipsel_mips32r2_glibc-2.27/mc-4.8.20/ipkg-install/opt/bin/mc /home/kaztost/Entware/build_dir/target-mipsel_mips32r2_glibc-2.27/mc-4.8.20/.pkgdir/mc/opt/bin
install -d -m0755 /home/kaztost/Entware/build_dir/target-mipsel_mips32r2_glibc-2.27/mc-4.8.20/.pkgdir/mc/opt/etc/mc
ln -sf mc /home/kaztost/Entware/build_dir/target-mipsel_mips32r2_glibc-2.27/mc-4.8.20/.pkgdir/mc/opt/bin/mcedit
cp -fpR /home/kaztost/Entware/build_dir/target-mipsel_mips32r2_glibc-2.27/mc-4.8.20/ipkg-install/opt/etc/mc/* /home/kaztost/Entware/build_dir/target-mipsel_mips32r2_glibc-2.27/mc-4.8.20/.pkgdir/mc/opt/etc/mc/
install -d -m0755 /home/kaztost/Entware/build_dir/target-mipsel_mips32r2_glibc-2.27/mc-4.8.20/.pkgdir/mc/opt/lib /home/kaztost/Entware/build_dir/target-mipsel_mips32r2_glibc-2.27/mc-4.8.20/.pkgdir/mc/opt/lib/mc /home/kaztost/Entware/build_dir/target-mipsel_mips32r2_glibc-2.27/mc-4.8.20/.pkgdir/mc/opt/lib/mc/extfs.d /home/kaztost/Entware/build_dir/target-mipsel_mips32r2_glibc-2.27/mc-4.8.20/.pkgdir/mc/opt/lib/mc/fish
cp -fpR /home/kaztost/Entware/build_dir/target-mipsel_mips32r2_glibc-2.27/mc-4.8.20/ipkg-install/opt/lib/mc/* /home/kaztost/Entware/build_dir/target-mipsel_mips32r2_glibc-2.27/mc-4.8.20/.pkgdir/mc/opt/lib/mc/
install -d -m0755 /home/kaztost/Entware/build_dir/target-mipsel_mips32r2_glibc-2.27/mc-4.8.20/.pkgdir/mc/opt/share /home/kaztost/Entware/build_dir/target-mipsel_mips32r2_glibc-2.27/mc-4.8.20/.pkgdir/mc/opt/share/mc /home/kaztost/Entware/build_dir/target-mipsel_mips32r2_glibc-2.27/mc-4.8.20/.pkgdir/mc/opt/share/mc/help /home/kaztost/Entware/build_dir/target-mipsel_mips32r2_glibc-2.27/mc-4.8.20/.pkgdir/mc/opt/share/mc/skins /home/kaztost/Entware/build_dir/target-mipsel_mips32r2_glibc-2.27/mc-4.8.20/.pkgdir/mc/opt/share/mc/hints /home/kaztost/Entware/build_dir/target-mipsel_mips32r2_glibc-2.27/mc-4.8.20/.pkgdir/mc/opt/share/mc/syntax
cp -fpR /home/kaztost/Entware/build_dir/target-mipsel_mips32r2_glibc-2.27/mc-4.8.20/ipkg-install/opt/share/mc/* /home/kaztost/Entware/build_dir/target-mipsel_mips32r2_glibc-2.27/mc-4.8.20/.pkgdir/mc/opt/share/mc/
/home/kaztost/Entware/staging_dir/host/bin/sed -i -e 's,/home/kaztost/Entware/staging_dir/host/bin,/opt/bin,g' /home/kaztost/Entware/build_dir/target-mipsel_mips32r2_glibc-2.27/mc-4.8.20/.pkgdir/mc/opt/lib/mc/ext.d/*
/home/kaztost/Entware/staging_dir/host/bin/sed -i -e 's,/home/kaztost/Entware/staging_dir/host/bin,/opt/bin,g' /home/kaztost/Entware/build_dir/target-mipsel_mips32r2_glibc-2.27/mc-4.8.20/.pkgdir/mc/opt/lib/mc/extfs.d/*
/home/kaztost/Entware/staging_dir/host/bin/sed: can't read /home/kaztost/Entware/build_dir/target-mipsel_mips32r2_glibc-2.27/mc-4.8.20/.pkgdir/mc/opt/lib/mc/extfs.d/*: No such file or directory
make[2]: *** [/home/kaztost/Entware/build_dir/target-mipsel_mips32r2_glibc-2.27/mc-4.8.20/.pkgdir/mc.installed] Error 2
make[2]: Leaving directory `/home/kaztost/Entware/feeds/packages/utils/mc'
make[1]: *** [package/feeds/packages/mc/compile] Error 2
make[1]: Leaving directory `/home/kaztost/Entware'
make: *** [package/mc/compile] Error 2
```
Manual copying missed files into /home/kaztost/Entware/build_dir/target-mipsel_mips32r2_glibc-2.27/mc-4.8.20/ipkg-install/opt/share/mc solved the problem

kaztost, 2018-03-22:
And in the installed mc the fish menu (Shell link, SFTP...) is empty! It seems something is broken there. Also mc was switched off in menuconfig from the beginning.

kaztost, 2018-03-22:
The problem  is almost closed. I enabled all features of mc in menuconfig and all problems went away. It is only unclear why mc is excluded in config by default for mipsel platform.

zyxmon, 2018-03-22:
mc is enabled - https://github.com/Entware/Entware/blob/master/configs/mipsel-3.4.config#L6031
Probably you have copied config before feeds were installed.

closed 2018-03-26 by kaztost