Entware/Entware #35: Package: dotnet

verysimplenick, 2018-03-27:
Package request
---
with net core runtime, we can run asp.net core sites and C # \ F # \ VB.net applications

- [x] No similar package already added and/or this one got some new killer feature,
- [x] It's not abandoned (at least for last 5 years or so),
- [ ] I successfully tested added package on my {platform}.

---
[binaries](https://github.com/dotnet/core-setup)
src can be found: https://github.com/dotnet/core/releases
[Maybe usefull](https://github.com/dotnet/coreclr/blob/master/Documentation/building/cross-building.md)

I tried download binaries for linux arm64 (`dotnet-runtime-latest-linux-arm64.tar.gz`), but can't run it.
`uname -a`

```
Linux qrouter 4.1.27 #2 SMP PREEMPT Sat Mar 10 14:37:32 EST 2018 aarch64 ASUSWRT-Merlin
```
`/tmp/mnt/sda1/dotnet# ./dotnet -v`

```
./dotnet: error while loading shared libraries: libstdc++.so.6: cannot open shared object file: No such file or directory

```

ryzhovau, 2018-03-27:
Try to install necessary libraries before running binaries.
```
opkg install libgcc libstdcpp
```

ryzhovau, 2018-03-27:
Ah! `dotnet` binary don't use `/opt/lib` libraries at all. Now try:
```
export LD_LIBRARY_PATH=/opt/lib:/opt/usr/lib:/lib:/usr/lib
./dotnet
```

zyxmon, 2018-03-27:
Third party binaries should use firmware libraries, not Entware ones. Try to take asuswrt-merlin sources and compile missing libs.

verysimplenick, 2018-03-27:
@ryzhovau now:
```
./dotnet -v
./dotnet: relocation error: /opt/lib/libc.so.6: symbol _dl_exception_create, version GLIBC_PRIVATE not defined in file ld-linux-aarch64.so.1 with link time reference
```
@zyxmon, if I understand right - you say build own asuswrt-merlin and include all dependencies in firmware dist, it's right?
p.s. можно и на русском, быстрее изъясняться будем, а потом в wiki выложим что получится
pps @zyxmon у вас на форуме я краем глаза видел довольно старый пост как собирать под entware (если честно пока не читал даже), может есть уже docker контейнер с уже установленными зависимостями для сборки пакетов под entware, может быстрее было бы собрать dotnet из исходников

zyxmon, 2018-03-27:
Можно не собирать всю прошивку мерлина. Вам пока не хватает libstdc++.so.6 и libgcc_s.so.1
Нужно, чтобы они появились при сборке. Это часть тулчейна (фыгыцке-merlin). Если они соберутся, но их поместить на роутер и задать LD_LIBRARY_PATH в папку, где они лежат.
Я не видел, чтобы удавалось запустить бинарники с библиотеками Entware, я бы и не пытался, хотя не исключаю, что иногда это возможно.
Пакет dotnet непростой, если нужен позарез, то способ запуска я привел. Все остальное сложнее в разы.

ryzhovau, 2018-03-27:
> Вам пока не хватает libstdc++.so.6 и libgcc_s.so.1
Нужно, чтобы они появились при сборке. Это часть тулчейна (asuswrt-merlin). 

В asuswrt-merlin тулчейн поставляется собранным, я сходуне вижу скриптов для того, чтобы его пересобрать. `libgcc_s.so` [есть](https://github.com/RMerl/am-toolchains/tree/15ddb6e7dad12bbb74de15f2a28319489b656a08/brcm-arm-hnd/crosstools-aarch64-gcc-5.3-linux-4.1-glibc-2.22-binutils-2.25/usr/aarch64-buildroot-linux-gnu/sysroot/lib), но в образ прошивки не пакуется, но `libstdc++.so` без пересборки никак не получить.

Так или иначе, остальные пути, включая кросс-компиляцию .NET Core, кратно сложнее.

verysimplenick, 2018-03-27:
Я видимо не совсем понимаю, обе so-шки по-сути лежат в системе
```
 ls -la /opt/lib/libstd* /opt/lib/libgcc_*
-rw-r--r--    1 admin    root         83960 Mar  3 14:05 /opt/lib/libgcc_s.so.1
lrwxrwxrwx    1 admin    root            19 Mar 27 02:03 /opt/lib/libstdc++.so.6 -> libstdc++.so.6.0.24
-rwxr-xr-x    1 admin    root       1710696 Mar  3 14:05 /opt/lib/libstdc++.so.6.0.24
-rw-r--r--    1 admin    root          2580 Mar  3 14:05 /opt/lib/libstdc++.so.6.0.24-gdb.py
```
Это собранные "entware"? А те что в `/lib/aarch64/` от Merlin-а и собственно поэтому не работает, так? Не понимаю пока почему нельзя указать что весь рантайм сишный взять из `/opt/lib/`. Подскажите где почитать.

ryzhovau, 2018-03-28:
Почитайте про `LD_LIBRARY_PATH`. В примере выше с экспортом этой переменной вы подсказали dynamic loader'у искать нужные библиотеки в `/opt/lib`. Библиотеки он нашёл, но нужных переменных\функций в них — нет:
```
./dotnet: relocation error: /opt/lib/libc.so.6: symbol _dl_exception_create, version GLIBC_PRIVATE not defined in file ld-linux-aarch64.so.1 with link time reference
```

Причина может быть в том, что наша Glibc собрана без каких-то фич, так необходимых бинарнику `dotnet`. 

Кстати говоря, а какая задача перед вами стоит? Какой консольный .NET-сервис, доступный исключительно в Windows-бинарниках и не использующий ничего за пределами .NET core вы собирались использовать? Я подступался к кросс-компиляции mono на пике популярности, но тогда игра свеч не стоила.

verysimplenick, 2018-03-28:
> Какой консольный .NET-сервис, доступный исключительно в Windows-бинарниках и не использующий ничего за пределами .NET core вы собирались использовать?

Всмысле windows-бинарниках? net core != net framework и он далеко не windows only, запускают даже на raspberry pi, кстати библиотеки gui тоже кросплатформенные third-party уже есть.
Вообще приобрел роутер asus rt-ac86u , ресурсов у него прилично и в идеале хотел запустить asp.net core сайт, да может пару сервисов в фон написать.
Только вот с прошивками, entware и роутерами дел не имел - опыта нет, вчера попробовал собрать merlin.ng , пока безуспешно. Меня смущает у него в мануале:

```PATH=/opt/toolchains/crosstools-arm-gcc-5.3-linux-4.1-glibc-2.22-binutils-2.25/usr/bin:/opt/toolchains/crosstools-aarch64-gcc-5.3-linux-4.1-glibc-2.22-binutils-2.25/usr/bin:/projects/hnd/tools/linux/hndtools-armeabi-2011.09/bin:$PATH```

** /projects/hnd/tools/linux/hndtools-armeabi-2011.09/bin ** вот это, откуда брать - непонятно, подозреваю что как раз из-за этого и не сбилдилось.

Однако я так и не понял, почему бы просто не пересобрать dotnet с рантаймом внутри типа
`-static-libgcc` и статически слинковать с `libstdc++.a` остальное то вроде находит, тогда и не предется мучаться с пересборкой asus-merlin.ng при выходе новых версий прошивки и т.д., разве не сработает так? Вопрос только в том как тулчейн развернуть под armv8 + те же версии других .so либ взять что у merlina используются.

ryzhovau, 2018-03-28:
Процесс сборки хорошо описан в wiki проекта asuswrt-merlin, помощь можно получить на [форуме SNB](https://www.snbforums.com/forums/asus-wireless.37/) или IRC-канале #asuswrt @dal.net ([Telegram<>IRC gate](https://t.me/asuswrt)). Здесь это выглядит откровенным офф-топиком.

В Entware добавлять .NET core, а уж тем более что-то за пределами Core необходимости не видим, извини.

closed 2018-03-28 by ryzhovau

verysimplenick, 2018-03-30:
Если вдруг кому интересно - hello world запустил на rt-ac86u (arm64v8).

dinobud, 2018-05-16:
@verysimplenick I am very interested in how you got this working. I was just trying to do the same thing with the same problems on the same router.

0x1000000, 2018-10-07:
@verysimplenick  I am also very interested in how you managed to run it.