Entware/Entware #36: ecryptfs-utils

g34rl00s3, 2018-03-27:
ecryptfs-utils (version 111)

My NAS supports the ecryptfs filesystem which is buildin into the kernel.
To be able to use this feature you'll need userland utilities which are provided by ecryptfs-utils

zyxmon, 2018-03-27:
What is wrong with build in encryption interface?
[image]

g34rl00s3, 2018-03-27:
The encrypted drive isn't the internal drive of the NAS itself but an external network drive.
Does this encrypt feature also work with mounted network shares on the NAS? I've so I could try to
migrate.

zyxmon, 2018-03-27:
There are other encrypted filesystems in Entware. May be they will be useful. They are fuse (userspace), not kernel space.

ryzhovau, 2018-04-08:
Closing for now.

closed 2018-04-08 by ryzhovau