Entware/Entware #37: amule won't restart in x64 repositories [Syonology] - Segmentation Fault

mattheweli, 2018-04-01:
As I reported in the title of this issue, I've installed Entware-3x/Entware on Synology DS218+ and I'm able to start when I freshly rebooted the NAS amule a first time. If I try to restart it by init.d commands, I got asegmentation fault error. I tried to see if there is some processes left with htop, but I discovered nothing.

I can say this is an issue related to x64 architecture, since I've faced it also with precedent version of entware-ng. In that case, installing x32 repo fixed the issue

I'm available to any further details you may need for fixing this

zyxmon, 2018-04-01:
I am afraid Entware developers do not use amule. Please run amuled from command line. Something like ` amuled -f -c /opt/var/amule -o` to see what is happening. Sometimes running program with strace shows what is wrong. Please investigate the case.

mattheweli, 2018-04-02:
I stopped and restarted amuled with the command above, but the only output I got is the following:
`Segmentation fault (core dumped)`
Is there a verbose command for amuled?

mattheweli, 2018-05-03:
I found a way to have more ingo on the problem (by using the restart function of the init script):

 Starting amuled... /opt/etc/init.d/rc.func: line 17:  3382 Segmentation fault      (core dumped) $PREARGS $PROC $ARGS > /dev/null 2>&1
             failed.

zyxmon, 2018-05-03:
Console log - 
```
# amuled -o -c /opt/var/amule
 2018-05-03 20:11:52: Logging to stdout enabled
 2018-05-03 20:11:52: Initialising aMuleD SVN compiled with wxBase v2.8.12
 2018-05-03 20:11:52: Checking if there is an instance already running...
 2018-05-03 20:11:52: No other instances are running.
!2018-05-03 20:11:52: ERROR: WARNING Warning! You are running aMule as root.
!2018-05-03 20:11:52: Doing so is not recommended for security reasons,
!2018-05-03 20:11:52: and you are advised to run aMule as an normal
!2018-05-03 20:11:52: user instead.

--------------------------------------------------
Warning! You are running aMule as root.
Doing so is not recommended for security reasons,
and you are advised to run aMule as an normal
user instead.
--------------------------------------------------

 2018-05-03 20:11:52: Creditfile loaded, 0 clients are known
 2018-05-03 20:11:52: *** TCP socket (ECServer) listening on 0.0.0.0:4712
 2018-05-03 20:11:52: Created Server UDP-Socket at port 4665
 2018-05-03 20:11:52: ListenSocket: Ok.
 2018-05-03 20:11:52: Created Client UDP-Socket at port 4672
 2018-05-03 20:11:52: Loading server.met file: /opt/var/amule/server.met
!2018-05-03 20:11:52: 6 servers in server.met found
 2018-05-03 20:11:52: Loading temp files from /opt/var/amule/Temp.
 2018-05-03 20:11:52: All PartFiles Loaded.
 2018-05-03 20:11:52: No part files found
 2018-05-03 20:11:52: No shareable files found in directory: /opt/var/amule/Incoming
 2018-05-03 20:11:52: Found 0 known shared files
 2018-05-03 20:11:52: amuled: OnInit - starting timer
 2018-05-03 20:11:52: Loading IP filters 'ipfilter.dat' and 'ipfilter_static.dat'.
 2018-05-03 20:11:52: Loaded 0 IP-ranges from '/opt/var/amule/ipfilter.dat'.
 2018-05-03 20:11:52: Loaded 0 IP-ranges from '/opt/var/amule/ipfilter_static.dat'.
 2018-05-03 20:11:52: IP filter is ready
!2018-05-03 20:11:52: Connecting
 2018-05-03 20:11:52: Connecting to TV Underground (176.103.48.36 - 176.103.48.36:4184) using protocol obfuscation.
!2018-05-03 20:11:52: No contacts found, please bootstrap, or download a nodes.dat file.
!2018-05-03 20:11:52: Kad started.
 2018-05-03 20:11:52: Connected to TV Underground (176.103.48.36:4184)
 2018-05-03 20:11:53: Connecting to GrupoTS Server (46.105.126.71 - 46.105.126.71:4661) using protocol obfuscation.
 2018-05-03 20:11:53: Connected to GrupoTS Server (46.105.126.71:4661)
 2018-05-03 20:11:53: Your copy of aMule is up to date.
 2018-05-03 20:12:01: WARNING: TV Underground (176.103.48.36:4184) - NG : You have a lowid. Please review your network config and/or your settings.
!2018-05-03 20:12:01: Connection established on: TV Underground
!2018-05-03 20:12:01: Connected to TV Underground with LowID
 2018-05-03 20:12:01: New clientid is 3365336
!2018-05-03 20:12:01: WARNING: You have received Low-ID!
 2018-05-03 20:12:01:   Most likely this is because you're behind a firewall or router.
 2018-05-03 20:12:01:   For more information, please refer to http://wiki.amule.org
 2018-05-03 20:12:01: ServerMessage: server version 17.15 (lugdunum)
 2018-05-03 20:12:01: ServerMessage: Please Visit http://safeandsoundhost.com for cheap and secure offshore shared/virtual/dedicated hosting
 2018-05-03 20:12:01: ServerMessage: This server is sponsored by TV Underground, so please visit https://tvunderground.org.ru
 2018-05-03 20:12:01: ServerMessage: For any questions or suggestions please visit the TV Underground forums
 2018-05-03 20:12:01: WARNING: TV Underground (176.103.48.36:4184) - NG : Please try to use a random port instead of 4662, because a lot of ISP have filters on 4662 port. Any number between 80 and 65530 should be OK. Thanks

```