Entware/Entware #38: Package request: zapret

Lichtenshtein, 2018-04-01:
Package request: zapret
[https://github.com/bol-van/zapret](url)
---
Bypass DPI in linux.
Bypassing the blocking of http and https web sites.

Platform Xiaomi-Mi Mini/Asus RT-N56U, mipsel

ryzhovau, 2018-09-12:
Did your firmware supports NFQUEUE target?
Be ready to turn off NAT hardware acceleration to test this utility.

Lichtenshtein, 2018-09-12:
I'm using Padavan Firmware. I don't have enough knowledge to answer. But, I believe, this module is not supported by this firmware, and I do not know how to add it.

```
/opt/home/admin # iptables -t mangle -I POSTROUTING -p tcp --dport 80 -j NFQUEUE --queue-num 200 --queue-bypass
iptables v1.4.16.3: unknown option "--queue-num"
Try `iptables -h' or 'iptables --help' for more information.

/opt/home/admin # modprobe -a xt_NFQUEUE
modprobe: module xt_NFQUEUE not found in modules.dep
```

ryzhovau, 2018-09-13:
Anyway, binaries are [ready](https://forum.keenetic.net/topic/3078-DPI-bypass/?do=findComment&comment=60927), now it's up to you.

closed 2018-09-27 by no-response[bot]