Entware/Entware #40: Pkg Request: Please update dnscrypt to the latest 2.0.8

ravenise, 2018-04-01:
Package request
---

ryzhovau, 2018-04-01:
dnscrypt 2.x was [added recently](https://github.com/Entware/entware-go/commit/60c2779a29fdd803ed2a956086cdca7ad565acb0).

ravenise, 2018-04-01:
Well, I found a work around: https://www.snbforums.com/threads/release-dnscrypt-installer-for-asuswrt.36071/

zyxmon, 2018-04-01:
>>  I do not see it in the opkg list and this is not up-to-date by default, 

Package will be available with the next sync. In fact binary is the same as github one.

ravenise, 2018-04-01:
Great! thank you so much.

nebularg, 2018-04-03:
@zyxmon how often is the sync? (just curious)

zyxmon, 2018-04-03:
We try to do it every month.

closed 2018-04-03 by zyxmon

ankar84, 2018-04-17:
Please update to [https://github.com/jedisct1/dnscrypt-proxy/releases/tag/2.0.10](2.0.10)
Changelog:

- Whitelists have been implemented: one a name matches a pattern in the whitelist, rules from the name-based and IP-based blacklists will be bypassed. Whitelists support the same patterns as blacklists, as
- well as time-based rules, so that some website can be normally blocked, but accessible on specific days or times of the day.
- Lists are now faster to load, and large lists require significantly less memory than before.
- New options have been added to disable TLS session tickets as well as use a specific cipher suite. See the example configuration file for a recommended configuration to speed up DoH servers on ARM such as Android devices and Raspberry Pi.
- The -service install command now remembers what the current directory was when the service was installed, in order to later load configuration files with relative paths.
- DoH: The "Cache-Control: max-age" header is now ignored.
- Patterns can now be prefixed with = to do exact matching: =example.com matches example.com but will not match www.example.com.
- Patterns are now fully supported by the cloaking module.
- A new option was added to use a specific cipher suite instead of the server's provided one. Using RSA+ChaChaPoly over ECDSA+AES-GCM has shown to decrease CPU usage and latency when connecting to Cloudflare, especially on Mips and ARM systems.
- The ephemeral keys mode of dnscrypt-proxy v1.x was reimplemented: this creates a new unique key for every single query.

Thanks!

ryzhovau, 2018-04-17:
One [issue](https://github.com/Entware/Entware/issues/54) is enough. Please, avoid cross-posting.

privum, 2018-05-02:
Hi,

What's the status on this? Would love to try it out.

Thanks

zyxmon, 2018-05-03:
test package (mipsel) is here
http://bin.entware.net/mipselsf-k3.4/test/dnscrypt-proxy2_2.0.11-1_mipsel-3.4.ipk

Quarx2k, 2018-05-12:
Works fine. Thanks!
Keenetic KN 1010

privum, 2018-05-14:
I'm confused. Has the update been added yet?

Because when I try to update it via entware it still shows the old package

Quarx2k, 2018-05-15:
@privum No, i just manually install ipk.

ryzhovau, 2018-05-15:
Binaries will arrive in couple of days.