Entware/Entware #41: Request to update Transmission to 2.93

minglespy, 2018-04-02:
Package request

versions earlier than 2.93 have been blocked by a number of trackers due to security flaws. Could you update the package to the latest version?

Many thanks.

zyxmon, 2018-04-02:
Transmission was updated to 2.93 two montes ago
https://github.com/Entware/entware-packages/blob/master/net/transmission/Makefile

closed 2018-04-02 by zyxmon

minglespy, 2018-04-02:
OK thanks - I was looking trying to upgrade a mipsel device rather than arm. The repository still has Transmission at 2.92 [http://pkg.entware.net/binaries/mipsel/Packages.html](url)

zyxmon, 2018-04-02:
@minglespy - you are using Entware-ng that will never be updated. If your kernel is 3.4 or higher - you'll need to upgrade to Entware. Mipsel devices with 2.6.22 kernel were a headache to adopt modern sources to compile and run. That is why Entware-ng development stopped.

minglespy, 2018-04-02:
@zyxmon - thanks for your help. I'm on using the ASUS AC66U on kernel 2.6.22 so it doesn't look like I'll be able to upgrade. Thanks anyway!