Entware/Entware #45: dnscrypt-proxy v2 DoH resolvers issue

ankar84, 2018-04-05:
Start [here](https://github.com/jedisct1/dnscrypt-proxy/issues/312)
Root cert issue is resolved now, but **CloudFlare DoH** resolver still not works for me.
Now I get that error:
[2018-04-05 18:57:31] [ERROR] Get https://dns.cloudflare.com/dns-query?body=yv4BAAABAAAAAAABAAACAAEAACkQAAAAgAAAAA&ct=&dns=yv4BAAABAAAAAAABAAACAAEAACkQAAAAgAAAAA&random_padding=XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX: **net/http: request canceled while waiting for connection (Client.Timeout exceeded while awaiting headers)**
[2018-04-05 18:57:31] [NOTICE] dnscrypt-proxy is waiting for at least one server to be reachable

I'm using **http://bin.entware.net/mipselsf-k3.4/test/dnscrypt-proxy2_2.0.8-1_mipsel-3.4.ipk** as @zyxmon  recommended.
For information: @jedisct1

zyxmon, 2018-04-05:
This is a question for @jedisct1

zyxmon, 2018-04-05:
It look like a duplicate of https://github.com/jedisct1/dnscrypt-proxy/issues/275

zyxmon, 2018-04-05:
It is useless do discuss it here. I am closing the issue. If you find the issue related to Entware (nonstandard paths as an example) - post here.

closed 2018-04-05 by zyxmon