Entware/Entware #50: Kernel 2.6.36 ipset not working

Kaol-Fu, 2018-04-09:
I have installed Entware and ipset successfully in my Asus router RT-AC68U. But when I use the ipset, it always echo the error message like the following.

Kaol@RT-AC68U:/tmp/home/root# uname -r
2.6.36.4brcmarm
Kaol@RT-AC68U:/tmp/home/root# opkg update ipset
Downloading http://bin.entware.net/armv7sf-k2.6/Packages.gz.
Updated list of available packages in /opt/var/opkg-lists/entware.
Kaol@RT-AC68U:/tmp/home/root# ipset create test hash:ip
ipset v6.34: Kernel error received: Invalid argument

ryzhovau, 2018-04-09:
Also, I don't know which ipset standart (v4 or newer) used by @RMerl.

Kaol-Fu, 2018-04-09:
I use the stock firmware not the merlin firmware. Could you please tell me how can I load the appropriate kernel modules first? Thanks!

ryzhovau, 2018-04-09:
Please, refer to asuswrt-merlin [wiki](https://github.com/RMerl/asuswrt-merlin/wiki/Legacy-Ipset-Scripts). I'm not sure you've got those modules on stock firmware.

zyxmon, 2018-04-09:
This is ipset legacy version compiled for this repo. It was used in Entware-ng.
http://bin.entware.net/armv7sf-k2.6/test/ipset4_4.5-1_armv7-2.6.ipk
Please try it instead of installed ipset that is version 6.

RMerl, 2018-04-09:
ipset modules are not part of the stock firmware, no.