Entware/Entware #54: Package request: dnscrypt-proxy v2

ankar84, 2018-04-17:
Package request: **dnscrypt-proxy v2.0.10**
https://github.com/jedisct1/dnscrypt-proxy
According to author of package "large lists require significantly less memory than before" in dnscrypt-proxy 2.0.8 which can be found at http://bin.entware.net/mipselsf-k3.4/test/dnscrypt-proxy2_2.0.8-1_mipsel-3.4.ipk
Thanks!
---
- Whitelists have been implemented: one a name matches a pattern in the whitelist, rules from the name-based and IP-based blacklists will be bypassed. Whitelists support the same patterns as blacklists, as
- well as time-based rules, so that some website can be normally blocked, but accessible on specific days or times of the day.
- Lists are now faster to load, and large lists require significantly less memory than before.
- New options have been added to disable TLS session tickets as well as use a specific cipher suite. See the example configuration file for a recommended configuration to speed up DoH servers on ARM such as Android devices and Raspberry Pi.
- The -service install command now remembers what the current directory was when the service was installed, in order to later load configuration files with relative paths.
- DoH: The "Cache-Control: max-age" header is now ignored.
- Patterns can now be prefixed with = to do exact matching: =example.com matches example.com but will not match www.example.com.
- Patterns are now fully supported by the cloaking module.
- A new option was added to use a specific cipher suite instead of the server's provided one. Using RSA+ChaChaPoly over ECDSA+AES-GCM has shown to decrease CPU usage and latency when connecting to Cloudflare, especially on Mips and ARM systems.
- The ephemeral keys mode of dnscrypt-proxy v1.x was reimplemented: this creates a new unique key for every single query.

ankar84, 2018-05-03:
Thanks a lot!
http://bin.entware.net/mipselsf-k3.4/test/dnscrypt-proxy2_2.0.11-1_mipsel-3.4.ipk

closed 2018-05-03 by ankar84

MyTimeKill, 2018-09-01:
Update please to the latest 2.0.16