Entware/Entware #56: Samba Update: Include VFS Recycle

auburnham, 2018-04-18:
Package request: Update Samba to include VFS Recycle module/library
---
{Describe killer feature of package: https://www.samba.org/samba/docs/current/man-html/vfs_recycle.8.html}

- [ X ] No similar package already added and/or this one got some new killer feature(s),
- [ X ] Package is in active development (at least, not abandoned for 5 years or more),
- [ ] I successfully tested added package on my {platform}.

---
Please, fill `{}` fields when asking for new package or delete this template in any other case.

zyxmon, 2018-04-19:
We compile Samba3x with options from Openwrt. We will add more options after options are added upstream.

closed 2018-04-19 by zyxmon

auburnham, 2018-04-19:
Thanks, I will follow up there.  Do you handle Samba4x independently, or does that also flow down from Openwrt?

zyxmon, 2018-04-19:
There is no Samba4 in openwrt.

auburnham, 2018-04-19:
Is this one maintained here? https://github.com/Entware/rtndev/tree/master/samba4x
If so, is it possible to include the VFS Recycle library in the Samba4x build?

zyxmon, 2018-04-20:
Yuo are welcome to make PR to add VFS Recycle to it.