Entware/Entware #58: Tvheadend Update

ajsomething, 2018-04-22:
Package request

Tvheadend Upgrade With incentives

- [ ] Package is not in active development (at least, abandoned for 2 years or more),

I am willing to put in the time to recompile the package and share but the process is so convoluted i`m at a loss where to start. With the new routers out they have tons of bandwidth available for recording live television. Synology is making Routers and NAS boxes with this functionality that cost over $1000 and do not include TV tuners.

The Posible Setup would be....
HDHomerun tuner ~ $150
DD-WRT compatable router with USB ~ $25
1TB USB storage drive ~ $40
 
This setup provides ALL devices attached to the router with Live HD TV streams. With the already supported VPN could provide these streams world wide. The major pitfall of all Entertainment platforms Plex, Kodi etc. is they require another computer as a server to host Tvheadend or another backend. 

Enough of the rant, 
What I need is, 
Entware Tvheadend package be recompiled with the build option --enable-hdhomerun_client.

what i'm willing to offer, 
A How To install Entware and Tvheadend tutorial, and a monetary financial contribution to the individual who can remake this package for me.

ryzhovau, 2018-09-11:
As for DVB-T/T2/C2, we trying to [develop](https://github.com/Entware/entware-dvb) this software for limited group of users because it requires some kernel modules. We offering tvheadend 4.2.6 for them as experimental version. I know tvheadend is just a EPG-driven IPTV DVR, but we should keep our repo firmware independent.

The main package for all of is still [tvheadend 4.0.10](https://github.com/Entware/entware-packages/tree/master/multimedia/tvheadend), taken from OpenWrt. The newer version requires more dependencies and consumes more CPU/RAM resources.

Bottom line: We won't update tvheadend, but we could try to build new version with `--enable-hdhomerun_client` option for your architecture. Please, get appropriate kernel modules to handle your DVB sticker and let us know you are ready.

Sorry for late answer.

closed 2018-09-25 by no-response[bot]

cuadedanh, 2019-01-07:
please build a update ver 4.2 (have auto scan m3u file) or send me source feeds or make file, i will build by my own. Thanh

cjd6568358, 2020-03-16:
@cuadedanh Can you send me a new version for 4.2 ?

cjd6568358, 2020-03-16:
@ryzhovau where is the new version?could you build a new ? it's 2020 years