Entware/Entware #61: Package / Patch request: Block AAAA in Dnsmasq

rugabunda, 2018-05-07:
I'd like to see a version of dnsmasq that filters AAAA requests altogether. There is a patch here for just that: https://gist.github.com/bearice/7d3dc0e63e003d752622

Reasons: Data exfiltration and command and control channels
https://isc.sans.edu/diary/Command+and+Control+Channels+Using+%22AAAA%22+DNS+Records/21301
Contrary to popular belief, there are some good reasons to disable IPv6/AAAA on a machine in a IPv4-only network, even where DNS works: Reduce broadcast load; Reduce load on DNS resolvers by almost 50%; Reduce connection start-up times (significantly where DNS caches are laggy); Follow best practices to disable non-functional features to enhance security and stability. 

If there is any way you can integrate this into Entware, that would be excellent and beneficial for many people.
Thank you!
---
{Describe killer feature of package}

- [ ] No similar package already added and/or this one got some new killer feature(s),
- [ ] Package is in active development (at least, not abandoned for 5 years or more),
- [ ] I successfully tested added package on my {platform}.

---
Please, fill `{}` fields when asking for new package or delete this template in any other case.

Magissia, 2018-05-15:
Won't it be easier to just disable IPv6 router side to achieve this ?

themiron, 2018-05-15:
IPv6-disabled host unlikely requests AAAA from dnsmasq and dual-stack hosts need AAAA obviously. There's 6-year old https://tools.ietf.org/html/rfc6555 implemented almost everywhere on clients., conection startup time is not affected, moreover no AAAA request responses or cached errors on broken DNS clients are harmful from increasing resolve delay to innacessible A+AAAA hosts impact.
Dnsmasq has small cache by default, search time is not causing significant delays, most of time is spent on network ops. It's possible to turn it off, but same with increase resolve latency in result.
Not sure about what was meant by broadcast load, dns protocol doesn't use broadcast. if you mean mdns, dnsmasq doesn't support it at all.
Proposed patch seems not valid, it breaks queries parsing on the first AAAA question with NOERROR status. This could cause cache poisoning effect on some clients, same as described in https://tools.ietf.org/html/rfc4074.
So, I would like to not accept this

closed 2018-05-15 by themiron