Entware/Entware #64: android 7.1.2 install error

qilong1980, 2018-05-11:
andriod 7.1.2,cpu info:
Processor       : AArch64 Processor rev 4 (aarch64)
processor       : 0
processor       : 1
processor       : 2
processor       : 3
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 
CPU implementer : 0x41
CPU architecture: AArch64
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : Amlogic

install entware(http://bin.entware.net/aarch64-k3.10/installer/generic.sh) and run /opt/bin/opkg error ,display "Segmentation fault "
i can try http://entware-3x.zyxmon.org/binaries/armv8/installer/install_android.sh,and The same mistake
what can i do ?

zyxmon, 2018-05-11:
What is your amlogic chipset? what is kernel version? What busybox/wget version was used to install?
Please note that busybox 2.26.2 downloads corrupted files! Upgrade it or use this one https://play.google.com/store/apps/details?id=ru.meefik.busybox
Please read the wiki - https://github.com/Entware/Entware/wiki/Install-on-Android 
Only alternative installer can be used on Android.

There is no single recipe to install Entware on Android, sometimes it may be tricky! I have successfully installed Entware (aarch64) on Amlogic S912. The only problem was busybox/wget version in my case.

qilong1980, 2018-05-11:
Thank you!
my cpu is amlogic S905,what problem was busybox/wget? i can download all files,but don't run it. i try update busybox.

zyxmon, 2018-05-11:
Please answer ALL questions before posting!