Entware/Entware #68: mipsel nginx is too old (legacy status)

Magissia, 2018-05-15:
Package request
---
{nginx is a well known web server and reverse proxy, reverse proxy is interesting here as you cannot NAT the same 443 port to multiple machines in IPv4

Upgrading to supported version will introduce lots of security fixes}

- [ ] No similar package already added and/or this one got some new killer feature(s),
- [X] Package is in active development (at least, not abandoned for 5 years or more),
- [ ] I successfully tested added package on my {platform}.

---

/ # opkg update
Downloading http://pkg.entware.net/binaries/mipsel/Packages.gz
Updated list of available packages in /opt/var/opkg-lists/entware-ng
/ # opkg list nginx
nginx - 1.12.1-1 - nginx is an HTTP and reverse proxy server, as well as a mail proxy server,
 written by Igor Sysoev.

zyxmon, 2018-05-15:
If your kernel is not too old, please, upgrade from Entware-ng to Entware. Entware-ng is not updated and supported.

closed 2018-05-15 by zyxmon

Magissia, 2018-05-15:
/ # opkg update
Downloading http://pkg.entware.net/binaries/mipsel/Packages.gz

Magissia, 2018-05-15:
Wiki says my device has it pre installed
https://github.com/Entware/Entware/wiki/Install-on-Padavan-firmware (This is Entware's wiki and not Entware-ng's wiki)

zyxmon, 2018-05-15:
Entware-ng will never be updated. Switch to Entware. 
Please read this - https://github.com/Entware/Entware/issues/67
You need to upgrade Padavan firmware and reinstall Entware.

Magissia, 2018-05-15:
/media/Main # opkg update
Downloading http://bin.entware.net/mipselsf-k3.4/Packages.gz
Updated list of available packages in /opt/var/opkg-lists/entware
/media/Main # opkg upgrade
/media/Main # opkg list nginx
nginx - 1.12.2-1 - nginx is an HTTP and reverse proxy server, as well as a mail proxy server,
 written by Igor Sysoev.