Entware/Entware #70: [Package request] Fail2ban 0.10 or superior (& deps)

Magissia, 2018-05-18:
Package request
---
[Fail2ban](https://github.com/fail2ban/fail2ban) is a known log analysis tool used to generate iptables rules according to logs content, perfect to restrict access to abuses sources dynamically

Fail2ban depends on python3 >3.2 (already available)
[gamin > 0.0.21](https://people.gnome.org/~veillard/gamin/index.html) or [pyinotify >0.8.3](https://github.com/seb-m/pyinotify)

deps cannot be installed using pip3 on some platform due to insufficient memory available for the deploy process (not as efficient as opkg)

- [X] No similar package already added and/or this one got some new killer feature(s),
- [X] Package is in active development (at least, not abandoned for 5 years or more),
- [ ] I successfully tested added package on my {platform}.

---

zyxmon, 2018-05-22:
@Magissia 
>> deps cannot be installed using pip3 on some platform due to insufficient memory available

Just add swap and try again.

closed 2018-06-03 by ryzhovau

NatoBoram, 2018-11-19:
There's also this answer on StackOverflow that might help : https://unix.stackexchange.com/a/377253/210455