Entware/Entware #71: [Package request] wireguard-go

maurerr, 2018-05-23:
Package request.

https://git.zx2c4.com/wireguard-go/about/
---
Wireguard:
- Simple & Easy-to-use
- Cryptographically Sound
- Minimal Attack Surface
- High Performance
- Well Defined & Thoroughly Considered
- there is no similar package for kernels >= 3.10
- Package is in active development

zyxmon, 2018-05-23:
We have recently added go compiler to Entware. Try to build it yourself and test if it works.

maurerr, 2018-05-24:
your go version doesn't work
root@mytomato-ea6400:~#  go version
Illegal instruction

zyxmon, 2018-05-24:
>> your go version doesn't work

What is your Entware feed? armv7-k2.6?

maurerr, 2018-05-24:
correct:
src/gz entware http://bin.entware.net/armv7sf-k2.6
should I reinstall ?

zyxmon, 2018-05-24:
>> should I reinstall ?

My guess the problem is your arm cpu does not have floating point. I'll try to build _nohf version later fro broadcom armv7.  go works for other hardware:

```
 # /opt/bin/go/bin/go version
go version go1.10.1 linux/arm
 # /opt/bin/go/bin/go run hello.go
hello world
 # /opt/bin/go/bin/go build hello.go
 # ./hello
hello world
# cat /opt/etc/entware_release
release=entware
arch=arm
cpu=cortex-a9
cpu_subtype=unknown
float=soft
gcc=7.3.0
gcc_flags=-O2 -pipe -march=armv7-a -mtune=cortex-a9 -fno-caller-saves -mfloat-abi=soft
libc=glibc
libc_version=2.27
 # cat hello.go
package main
import "fmt"
func main() {
    fmt.Println("hello world")
}

```

zyxmon, 2018-05-24:
@maurerr - please uninstall go and install go_nohf instead of it.
I have tested it on broadcom armv7 (tomato) and it works.

maurerr, 2018-05-24:
it worked ! thanks !
now I'm trying to build the wireguard-go client

0x5e, 2020-05-18:
https://github.com/Entware/Entware/issues/453