Entware/Entware #72: [Package request] qbittorrent

glolsh, 2018-05-23:
Package request
---
GNU GPL bittorrent client with web interface and huge sorting functions, like tags and categories and sub-categories

- [ ] No similar package already added and/or this one got some new killer feature(s),
- [x] Package is in active development (at least, not abandoned for 5 years or more),

---

This package was added in qts 4.3 but my NAS stopped updating at 4.2.6 so I think that makes it more desirable.

zyxmon, 2018-05-23:
It hat QT as dependancy. It is heavy for embedded devices. Entware is used mainly on routers, not NASes.
We are not planning to add it in near future.

closed 2018-05-23 by zyxmon

glolsh, 2018-05-23:
Okay, thank you!

zyxmon, 2019-02-01:
qt5 and qbittorent are now in repo.

DanPM, 2019-05-12:
> qt5 and qbittorent are now in repo.

can't find in any repository the qbittorrent? Am I'm missing something?
Tried opkg install qbittorrent and opkg install qbittorrent-nox.
I have entware-ng0.97 on my qnap tvs-673

ryzhovau, 2019-05-12:
https://github.com/Entware/Entware#entware-ng-3x-and-entware-ng-merged-to-become-entware