Entware/Entware #73: MariaDB-Server 10.1.32 Cannot Initialize

sunjingwei789, 2018-05-28:
admin@RT-AC3100-AF2B:/tmp/mnt/sdb1/entware/root# ./bin/mysql_install_db --force
-sh: ./bin/mysql_install_db: not found
admin@RT-AC3100-AF2B:/tmp/mnt/sdb1/entware/root# /opt/bin/mysql_install_db --for
ce

FATAL ERROR: Could not find ./bin/my_print_defaults

If you compiled from source, you need to either run 'make install' to
copy the software into the correct location ready for operation.
If you don't want to do a full install, you can use the --srcddir
option to only install the mysql database and privilege tables

If you are using a binary release, you must either be at the top
level of the extracted archive, or pass the --basedir option
pointing to that location.

The latest information about mysql_install_db is available at
https://mariadb.com/kb/en/installing-system-tables-mysql_install_db

zyxmon, 2018-05-28:
Please run installation with
`mysql_install_db --basedir=/opt`
We will modify installation script with next package build

closed 2018-05-28 by zyxmon

sunjingwei789, 2018-05-29:
admin@RT-AC3100-AF2B:/tmp/home/root# cd /opt/root
admin@RT-AC3100-AF2B:/tmp/mnt/sdb1/entware/root# mysql_install_db --basedir=/opt
chown: unknown user root
Cannot change ownership of the database directories to the 'root'
user.  Check that you have the necessary permissions and try again.

sunjingwei789, 2018-05-29:
admin@RT-AC3100-AF2B:/tmp/mnt/sdb1/entware/root# mysql_install_db --basedir=/opt
chown: unknown user root
Cannot change ownership of the database directories to the 'root'
user.  Check that you have the necessary permissions and try again.

sunjingwei789, 2018-05-29:
admin@RT-AC3100-AF2B:/tmp/home/root# mysql_install_db --basedir=/opt
chown: unknown user root
Cannot change ownership of the database directories to the 'root'
user.  Check that you have the necessary permissions and try again.

sunjingwei789, 2018-05-29:
This version is basedir in /opt/etc/mysql/my.cnf = /opt
The problem is not here

sunjingwei789, 2018-05-29:
Hope to test

sunjingwei789, 2018-05-29:
already solved
cd /opt
./bin/mysql_install_db --force

sunjingwei789, 2018-05-29:
My account name is admin (firmware cannot set user name to root)
I modified user= admin in /opt/etc/mysql/my.cnf file
But there was a mistake. Ask, do you still need to make other changes?
Incidental, wrong text
Admin@RT-AC3100-AF2B:/tmp/home/root# cd /opt
Admin@RT-AC3100-AF2B:/tmp/mnt/sdb1/entware# ./bin/mysql_install_db –force
Installing MariaDB/MySQL system tables in '/opt/var/mysql/' ...
2018-05-29 5:48:35 716386848 [Note] Using unique option prefix 'key_buffer' is error-prone and can break in the future. Please use the full name 'key_buffer_size' instead.
2018-05-29 5:48:35 716386848 [Note] / opt / bin / mysqld (mysqld 10.1.32-MariaDB) starting as process 7573 ...
/opt/bin/mysqld: Can't change to run as user 'admin'. Please check that the user exists!
2018-05-29 5:48:35 716386848 [Note] InnoDB: innodb_empty_free_list_algorithm has been changed to legacy because of small buffer pool size. In order to use backoff, increase buffer pool at least up to 20MB.

2018-05-29 5:48:35 716386848 [Note] InnoDB: Using mutexes to ref count buffer pool pages
2018-05-29 5:48:35 716386848 [Note] InnoDB: The InnoDB memory heap is disabled
2018-05-29 5:48:35 716386848 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
2018-05-29 5:48:35 716386848 [Note] InnoDB: GCC builtin __atomic_thread_fence() is used for memory barrier
2018-05-29 5:48:35 716386848 [Note] InnoDB: Compressed tables use zlib 1.2.11
2018-05-29 5:48:35 716386848 [Note] InnoDB: Using generic crc32 instructions
2018-05-29 5:48:35 716386848 [Note] InnoDB: Initializing buffer pool, size = 128.0M
InnoDB: mmap(138543104 bytes) failed; errno 12
2018-05-29 5:48:35 716386848 [ERROR] InnoDB: Cannot allocate memory for the buffer pool
2018-05-29 5:48:35 716386848 [ERROR] Plugin 'InnoDB' init function returned error.
2018-05-29 5:48:35 716386848 [ERROR] Plugin 'InnoDB' registration as a STORAGE ENGINE failed.
/opt/bin/mysqld: Too many arguments (first extra is '–force').
2018-05-29 5:48:35 716386848 [ERROR] Aborting

Installation of system tables failed! Examine the logs in
/opt/var/mysql/ for more information.

The problem could be conflicting information in an external
My.cnf files. You can ignore these by doing:

    Shell> ./bin/mysql_install_db --defaults-file=~/.my.cnf

You can also try to start the mysqld daemon with:

    Shell> /opt/bin/mysqld --skip-grant-tables --general-log &

And use the command line tool /opt/bin/mysql
To connect to the mysql database and look at the grant tables:

    Shell> /opt/bin/mysql -u root mysql
    Mysql> show tables;

Try 'mysqld --help' if you have problems with paths. Using
--general-log gives you a log in /opt/var/mysql/ that may be helpful.

The latest information about mysql_install_db is available at
Https://mariadb.com/kb/en/installing-system-tables-mysql_install_db
You can find the latest source at https://downloads.mariadb.org and
The maria-discuss email list at https://launchpad.net/~maria-discuss

Please check all of the above before submitting a bug report
At http://mariadb.org/jira

sunjingwei789, 2018-05-29:
My account name is admin (firmware cannot set user name to root)
I modified user= admin in /opt/etc/mysql/my.cnf file
But there was a mistake. Ask, do you still need to make other changes?
Incidental, wrong text
admin@RT-AC3100-AF2B:/tmp/home/root# cd /opt
admin@RT-AC3100-AF2B:/tmp/mnt/sdb1/entware# ./bin/mysql_install_db –force
Installing MariaDB/MySQL system tables in '/opt/var/mysql/' ...
2018-05-29 5:48:35 716386848 [Note] Using unique option prefix 'key_buffer' is error-prone and can break in the future. Please use the full name 'key_buffer_size' instead.
2018-05-29 5:48:35 716386848 [Note] / opt / bin / mysqld (mysqld 10.1.32-MariaDB) starting as process 7573 ...
/opt/bin/mysqld: Can't change to run as user 'admin'. Please check that the user exists!
2018-05-29 5:48:35 716386848 [Note] InnoDB: innodb_empty_free_list_algorithm has been changed to legacy because of small buffer pool size. In order to use backoff, increase buffer pool at least up to 20MB.

2018-05-29 5:48:35 716386848 [Note] InnoDB: Using mutexes to ref count buffer pool pages
2018-05-29 5:48:35 716386848 [Note] InnoDB: The InnoDB memory heap is disabled
2018-05-29 5:48:35 716386848 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
2018-05-29 5:48:35 716386848 [Note] InnoDB: GCC builtin __atomic_thread_fence() is used for memory barrier
2018-05-29 5:48:35 716386848 [Note] InnoDB: Compressed tables use zlib 1.2.11
2018-05-29 5:48:35 716386848 [Note] InnoDB: Using generic crc32 instructions
2018-05-29 5:48:35 716386848 [Note] InnoDB: Initializing buffer pool, size = 128.0M
InnoDB: mmap(138543104 bytes) failed; errno 12
2018-05-29 5:48:35 716386848 [ERROR] InnoDB: Cannot allocate memory for the buffer pool
2018-05-29 5:48:35 716386848 [ERROR] Plugin 'InnoDB' init function returned error.
2018-05-29 5:48:35 716386848 [ERROR] Plugin 'InnoDB' registration as a STORAGE ENGINE failed.
/opt/bin/mysqld: Too many arguments (first extra is '–force').
2018-05-29 5:48:35 716386848 [ERROR] Aborting

Installation of system tables failed! Examine the logs in
/opt/var/mysql/ for more information.

The problem could be conflicting information in an external
My.cnf files. You can ignore these by doing:

    Shell> ./bin/mysql_install_db --defaults-file=~/.my.cnf

You can also try to start the mysqld daemon with:

    Shell> /opt/bin/mysqld --skip-grant-tables --general-log &

And use the command line tool /opt/bin/mysql
To connect to the mysql database and look at the grant tables:

    Shell> /opt/bin/mysql -u root mysql
    Mysql> show tables;

Try 'mysqld --help' if you have problems with paths. Using
--general-log gives you a log in /opt/var/mysql/ that may be helpful.

The latest information about mysql_install_db is available at
Https://mariadb.com/kb/en/installing-system-tables-mysql_install_db
You can find the latest source at https://downloads.mariadb.org and
The maria-discuss email list at https://launchpad.net/~maria-discuss

Please check all of the above before submitting a bug report
At http://mariadb.org/jira

sunjingwei789, 2018-05-29:
After an error occurs, many code returns -sh: /: Permission denied
/opt/bin/mysql
/opt/bin/mysql -u root mysql/opt/bin/mysqld --skip-grant-tables
--general-log&
And ./bin/mysql_install_db --defaults-file =~/ .my.cnf is there and
Cd /opt
./bin/mysql_install_db --force error
Just died on the permission card
My user group is root and the user name is admin

sunjingwei789, 2018-05-29:
Many code returns -sh: /: Permission denied after an error
E.g
/opt/bin/mysql
/opt/bin/mysql -u root mysql/opt/bin/mysqld --skip-grant-tables
--general-log&
And ./bin/mysql_install_db --defaults-file =~/ .my.cnf is there and
Cd /opt
./bin/mysql_install_db --force error
Just died on the permission card
My user group is root and the user name is admin

sunjingwei789, 2018-05-29:
After an error occurs, many code returns -sh: /: Permission denied
/opt/bin/mysql
/opt/bin/mysql -u root mysql/opt/bin/mysqld --skip-grant-tables
--general-log&
And ./bin/mysql_install_db --defaults-file =~/ .my.cnf is there and
Cd /opt
./bin/mysql_install_db --force error
Just died in the permissions

zyxmon, 2018-05-29:
The simplest temporary workaround is to use Entware alternative install - in this case you will have root account independent from your firmware.
We will investigate the issue with admin superuser.

omega69, 2018-05-31:
**Wanted to install Maria-db as well. Few failures but in the end seems it works**

1) root changed to admin in /opt/etc/mysql/my.cnf file
2) cd /opt
3) ./bin/mysql_install_db

**So I have not used "force" but this installation seems succesful. But there are other issues after**

**Cannot set password or start secure installation. Lines below are directly after successful installation**

> PLEASE REMEMBER TO SET A PASSWORD FOR THE MariaDB root USER !
> To do so, start the server, then issue the following commands:
> 
> '/opt/bin/mysqladmin' -u root password 'new-password'
> '/opt/bin/mysqladmin' -u root -h RT-AC56U password 'new-password'
> 
> Alternatively you can run:
> '/opt/bin/mysql_secure_installation'

admin@RT-AC56U:/tmp/mnt/sdb1/entware/bin# ps | grep sql
23572 admin     517m S    /opt/bin/mysqld --pid-file=/opt/var/run/mysqld.pid

admin@RT-AC56U:/tmp/mnt/sdb1/entware/bin# ls -al | grep mysql
-rwxr-xr-x    1 admin    root         18837 May 13 09:55 mysql_install_db
-rwxr-xr-x    1 admin    root        931280 May 13 09:55 mysql_upgrade
-rwxr-xr-x    1 admin    root       9481548 May 13 09:55 mysqld

**mysqladmin neither mysql_secure_installation are available in /opt/bin/**

admin@RT-AC56U:/tmp/mnt/sdb1/entware/bin# '/opt/bin/mysqladmin' -u root -h RT-AC56U password XXXXXXXXX
-sh: /opt/bin/mysqladmin: not foundadmin@RT-AC56U:/tmp/mnt/sdb1/entware/bin# '/opt/bin/mysqladmin' -u root password XXXXXXXXX
-sh: /opt/bin/mysqladmin: not found
admin@RT-AC56U:/tmp/mnt/sdb1/entware/bin# '/opt/bin/mysqladmin' -u admin password XXXXXXXXX
-sh: /opt/bin/mysqladmin: not found
admin@RT-AC56U:/tmp/mnt/sdb1/entware/bin# '/opt/bin/mysqladmin' -u admin -h RT-AC56U password XXXXXXXXX
-sh: /opt/bin/mysqladmin: not found
admin@RT-AC56U:/tmp/mnt/sdb1/entware/bin# /opt/bin/mysql_secure_installation
-sh: /opt/bin/mysql_secure_installation: not found

**latest RMerlin, AC56U, entware installed today**

zyxmon, 2018-05-31:
mariadb consist of 40+ packages (this is openwrt choice we follow). mysqlmyadmin is in mariadb-client-extra package. We are still fixing mariadb in Entware. We will report here when the new build is available.

omega69, 2018-05-31:
with the mariadb-clien-extra adding of users works well. phpmyadmin is now up as well

zyxmon, 2018-06-02:
mariadb updated to 10.1.33 and fixed.

ryzhovau, 2018-06-02:
https://github.com/Entware/entware-packages/commit/8e34a78df42acf093a69607e680ab242dbd00107