Entware/Entware #74: Entware fails installation - Package zoneinfo-asia md5sum mismatch

WillyTP, 2018-05-28:
Hello,
I'm reporting a bug regarding this package.

When I try to install Entware on my Asus RT-AC68U device, I get the following error which results in a not working installation.

`Configuring entware-release.
Collected errors:
 * opkg_install_pkg: Package zoneinfo-asia md5sum mismatch. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package entware-opt.
cp: can't stat '/opt/etc/shells.1': No such file or directory
Info: Congratulations!
Info: If there are no errors above then Entware was successfully initialized.
Info: Add /opt/bin & /opt/sbin to your PATH variable
Info: Add '/opt/etc/init.d/rc.unslung start' to startup script for Entware services to start
Info: Found a Bug? Please report at https://github.com/Entware/Entware/issues
`

zyxmon, 2018-05-29:
What installation is used to install Entware? What is your firmware? What is the filesystem on your usb drive. May be your ISP uses transparent proxy?
I have checked the repo - it is has correct checksums
```
~ # opkg update
Downloading http://bin.entware.net/armv7sf-k2.6/Packages.gz
Updated list of available packages in /opt/var/opkg-lists/entware
.....
~ # opkg install zoneinfo-asia
Installing zoneinfo-asia (2018e-1) to root...
Downloading http://bin.entware.net/armv7sf-k2.6/zoneinfo-asia_2018e-1_armv7-2.6.ipk
Configuring zoneinfo-asia.

```

WillyTP, 2018-05-29:
Hi Zyxmon
I'm trying to install Entware an a Asus RT-AC68U with latest Merlin 384.5 firmware;
Installation media is a USB Key with Ext4 - no journal file system.
I'm connecting through a mobile LTE connection, for being exact with Vodafone.

I tried your steps without success.
Sounds so strange to me, other packages, other things I've installed seems to work properly.
```
ASUSWRT-Merlin RT-AC68U 384.5-0 Sun May 13 01:53:17 UTC 2018
WillyTP@RT-AC68U-WMPP:/tmp/home/root# opkg update
Downloading http://bin.entware.net/armv7sf-k2.6/Packages.gz
Updated list of available packages in /opt/var/opkg-lists/entware
WillyTP@RT-AC68U-WMPP:/tmp/home/root# opkg install zoneinfo-asia
Installing zoneinfo-asia (2018e-1) to root...
Downloading http://bin.entware.net/armv7sf-k2.6/zoneinfo-asia_2018e-1_armv7-2.6.ipk
Collected errors:
 * opkg_install_pkg: Package zoneinfo-asia md5sum mismatch. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package zoneinfo-asia.
```

```
WillyTP@RT-AC68U-WMPP:/tmp/home/root# wget http://bin.entware.net/armv7sf-k2.6/zoneinfo-asia_2018e-1_armv7-2.6.ipk
--2018-05-29 09:36:45--  http://bin.entware.net/armv7sf-k2.6/zoneinfo-asia_2018e-1_armv7-2.6.ipk
Resolving bin.entware.net... 81.4.123.217
Connecting to bin.entware.net|81.4.123.217|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/x-gzip]
Saving to: 'zoneinfo-asia_2018e-1_armv7-2.6.ipk'

zoneinfo-asia_2018e-1_armv7-2.6     [ <=>                                                   ]  29.90K  --.-KB/s    in 0.08s   

2018-05-29 09:36:45 (397 KB/s) - 'zoneinfo-asia_2018e-1_armv7-2.6.ipk' saved [30622]

WillyTP@RT-AC68U-WMPP:/tmp/home/root# md5sum zoneinfo-asia_2018e-1_armv7-2.6.ipk 
dc78e72b45bff2b51f9fb37ca3ecd2bc  zoneinfo-asia_2018e-1_armv7-2.6.ipk
WillyTP@RT-AC68U-WMPP:/tmp/home/root# opkg install /tmp/home/root/zoneinfo-asia_2018e-1_armv7-2.6.ipk 
Collected errors:
 * pkg_init_from_file: Malformed package file /tmp/home/root/zoneinfo-asia_2018e-1_armv7-2.6.ipk.
```

WillyTP, 2018-05-29:
Update:
I tried to download zoneinfo-asia via HTTPS, it works and md5sum is different:

```
WillyTP@RT-AC68U-WMPP:/tmp/home/root# wget https://bin.entware.net/armv7sf-k2.6/zoneinfo-asia_2018e-1_armv7-2.6.ipk
--2018-05-29 10:19:44--  https://bin.entware.net/armv7sf-k2.6/zoneinfo-asia_2018e-1_armv7-2.6.ipk
Resolving bin.entware.net... 81.4.123.217
Connecting to bin.entware.net|81.4.123.217|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 30497 (30K) [application/x-gzip]
Saving to: 'zoneinfo-asia_2018e-1_armv7-2.6.ipk'

zoneinfo-asia_2018e-1_armv7-2.6 100%[======================================================>]  29.78K  --.-KB/s    in 0.005s  

2018-05-29 10:19:45 (5.98 MB/s) - 'zoneinfo-asia_2018e-1_armv7-2.6.ipk' saved [30497/30497]

WillyTP@RT-AC68U-WMPP:/tmp/home/root# md5sum zoneinfo-asia_2018e-1_armv7-2.6.ipk 
6c612c4711d486be8b217892e25f4233  zoneinfo-asia_2018e-1_armv7-2.6.ipk
WillyTP@RT-AC68U-WMPP:/tmp/home/root# opkg install /tmp/home/root/zoneinfo-asia_2018e-1_armv7-2.6.ipk 
Installing zoneinfo-asia (2018e-1) to root...
Configuring zoneinfo-asia.
```

Is it possibile to force HTTPS to download packages?

zyxmon, 2018-05-29:
>> Is it possibile to force HTTPS to download packages?
opkg binary calls wget to download - and it does not support https.
Probably we will add https in the future.

closed 2018-05-29 by zyxmon

ryzhovau, 2018-05-29:
> Is it possibile to force HTTPS to download packages?

https://github.com/Entware/Entware/wiki/Using-HTTPS-for-opkg

WillyTP, 2018-05-29:
Hi,
I can make the change posted by ryzhovau,
however if I try to run the script from scratch it fails.

For the sake of knowledge, what do you think is causing this weird problem to me?

Thank you!

zyxmon, 2018-05-29:
>> I can make the change posted by ryzhovau

It is a change after Entware is installed. 
There may be some problem with your ISP that cases corrupted downloads or old downloads (transparent proxy).

WillyTP, 2018-06-01:
Hello everybody.
Well, I think zyxmon was right.

I tried to setup a VPN before, and everything went smooth.

Hope this helps other people experiencing the same issue.