Entware/Entware #75: help:MariaDB-Server 10.1.32 Cannot Initialize

sunjingwei789, 2018-05-29:
[image]
[image]
After an error occurs, many code returns -sh: /: Permission denied
/opt/bin/mysql
/opt/bin/mysql -u root mysql/opt/bin/mysqld --skip-grant-tables
--general-log&
And ./bin/mysql_install_db --defaults-file =~/ .my.cnf is there and
Cd /opt
./bin/mysql_install_db --force error
Just died in the permissions

sunjingwei789, 2018-05-29:
Cannot set user name to root due to router
You can only debug with the admin account, but it belongs to the root user group.
Has been user.admin in my.cnf, the above error

sunjingwei789, 2018-05-29:
Cannot set user name to root due to router
I can only debug with the admin account, but it belongs to the root user group.
Has been user.admin in my.cnf, the above error

sunjingwei789, 2018-05-29:
Cannot set user name to root due to router
I can only debug with the admin account, but it belongs to the root user group.
Has been user=admin in my.cnf, the above error

sunjingwei789, 2018-05-29:
I have put user=admin in my.cnf
Cannot set user name to root due to router
I can only debug with the admin account, but it belongs to the root user group.

zyxmon, 2018-05-29:
The same as  https://github.com/Entware/Entware/issues/73

closed 2018-05-29 by zyxmon