Entware/Entware #76: Package request: MTProxy

Kirya2k, 2018-05-30:
Package request: MTProxy

Official proxy of Telegram messenger.
https://github.com/TelegramMessenger/MTProxy/

So, please, add the package with capabilities for mipsel platform, thanks a lot.
---
- [V] No similar package already added and this one got some new killer feature(s),
- [V] Package is in active development 
- [  ]  I successfully tested added package on my {platform}.

ryzhovau, 2018-05-31:
Unfortunately, `MPTproxy` is not ready for cross-compilation. It uses specific [asm instructions](https://github.com/TelegramMessenger/MTProxy/blob/cc7b7097a5bc1681d0f4c05d5f07576c8b279fc4/common/crc32c.c#L305) and other x86 optimizations.

Also, it uses newer openssl 1.1.x while Entware packages relies on 1.0.x.

ryzhovau, 2018-06-26:
Referring to https://github.com/TelegramMessenger/MTProxy/issues/18

ryzhovau, 2019-01-23:
There's some [patches](https://github.com/openwrt/packages/pull/7940/files) to avoid x86 specific assembler instructions.

ryzhovau, 2019-01-23:
Still not usable, going to sit on this and wait forawhile
```
~ # mtproto-proxy --help
mtproto-proxy: engine/engine.c:576: engine_startup: Assertion `SIGRTMAX == OUR_S IGRTMAX' failed.
[pid 17582] [time 1548236193]
------- Stack Backtrace -------
mtproto-proxy(print_backtrace+0x34)[0x44ba08]
[pid 17582] [time 1548236193] -------------------------------
[pid 17582] [time 1548236193] mtproxy-0.01 compiled at Jan 22 2019 08:22:50 by g cc 7.3.0 32-bit after commit 1[pid 17582] [time 1548236193]
```

ryzhovau, 2019-01-23:
Please, check this build, where this assertion check commented out. 
```
~ # opkg install xxd curl
~ # opkg install http://bin.entware.net/mipselsf-k3.4/archive/mtproxy_1-1_mipsel-3.4.ipk
~ # head -c 16 /dev/urandom | xxd -ps
XXXXXXXXXXXXXXX
~ # curl https://core.telegram.org/getProxyConfig > proxy-multi.conf
~ # curl https://core.telegram.org/getProxySecret > proxy-secret
~ # mtproto-proxy -u nobody -p 8888 -H 444 -S XXXXXXXXXXXXXXX --aes-pwd proxy-secret proxy-multi.conf -M 1
```

themiron, 2019-01-23:
fyi, this app is quite cpu-hungry
refer https://github.com/TelegramMessenger/MTProxy/issues/100

Kirya2k, 2019-01-23:
Working.
And memory hungry too.
:(

root@Keenetic_Giga ~# ps aux --sort -rss
USER       PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
...
nobody    8920  6.3  1.2 253084  3076 pts/2    Sl+  17:47   0:22 mtproto-proxy -

closed 2019-01-23 by ryzhovau

themiron, 2019-01-23:
at the same time...
```
USER       PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
admin     1876  0.2  2.9  20404 13108 pts/1    S+   20:08   0:00 python3 ./mtprotoproxy.py
```