Entware/Entware #77: Samba server cannot handle files over 2GB

gadget1999, 2018-06-02:
I'm using the latest Samba Server 4.8.0 package (http://bin.entware.net/mipselsf-k3.4/samba4x-server_4.8.0-1_mipsel-3.4.ipk) on Asus N56U router (Linux kernel: 3.4.113). An external 2TB USB hard drive is used as the file share drive, it's formatted into ext4.

So far everything is working perfectly except one thing: any files larger than 2GB are not showing up on SMB clients (no matter it's Linux or Windows), and creating a file larger than 2GB will fail after 2GB is copied. I tried to use vsftpd to share files on the same hard drive, it works fine with files larger than 2GB. Trying to read/write large files on the USB hard drive also worked fine.

From the smbd debug logging, it appears the input offset will become negative after 2GB mark. The output is enclosed. It sounds a bug of Samba server? Thanks a lot for the help!

[smbd-output.log](https://github.com/Entware/Entware/files/2065446/smbd-output.log)

------- excerpt of output --------------
smbd_smb2_write: Downloads/2gb - fnum 2357515247
smb2: write size (1048576) too small for minimum aio_write of 0
real_write_file (Downloads/2gb): **pos = -2118123520**, size = 1048576, returned -1
smb2_write failed: fnum 2357515247, file Downloads/2gb, length=1048576 offset=0 nwritten=-1: **NT_STATUS_INVALID_PARAMETER**
smb2: write on file Downloads/2gb, offset 2176843776, requested 1048576, written = 4294967295
smbd_smb2_request_error_ex: smbd_smb2_request_error_ex: idx[1] status[NT_STATUS_INVALID_PARAMETER] || at **../source3/smbd/smb2_write.c:135**
smbd_smb2_request_done_ex: idx[1] status[NT_STATUS_INVALID_PARAMETER] body[8] dyn[yes:1] at ../source3/smbd/smb2_server.c:3200
smb2_set_operation_credit: requested 16, charge 16, granted 16, current possible/max 16/512, total granted/max/low/range 512/8192/33267/512

gadget1999, 2018-06-03:
Some additional info about this:

I tried to install the latest samba36-server package (ver 3.6.25), files larger than 2GB are working fine. However, I could not use SMB3 on it, which was the reason I used Samba 4.8.0.

I also tried to use SMB1.0 to connect to Samba 4.8.0 (just to test), 2GB files are not working either.

ryzhovau, 2018-09-12:
Samba4 now is a [part](https://github.com/openwrt/packages/tree/master/net/samba4) of OpenWrt feed.
It was re-worked by new maintainers. Please, give them a [note](https://github.com/openwrt/packages/issues) if something wrong for now.

closed 2018-09-12 by ryzhovau