Entware/Entware #78: duktape javascript broken

bayonne, 2018-06-04:
Package duktape provides a javascript engine.
Javascript session:

```
duk> print('a')
ReferenceError: identifier 'print' undefined
    at [anon] (duk_js_var.c:1198) internal
    at global (input:1) preventsyield

```
Any idea how fix that ?

zyxmon, 2018-06-04:
http://duktape.org/guide.html#compiling.2.1
Probably default configuration has "No debug printing, no asserts, etc".
Probably @ryzhovau  will be more specific.

zyxmon, 2018-06-04:
@ryzhovau - I think we should leave libduktape in binary repo for gerbera and remove executables.

ryzhovau, 2018-06-04:
Also, it's an optional dep for `pyload` as the only modern JS-engine in our repo.
I'll look what else I can do.

bayonne, 2018-06-04:
Thank you. Sorry to have called the package broken.
I come from `pyload` :)

ryzhovau, 2018-06-04:
Looks like `print()` is [no longer automatically provided](https://github.com/svaarala/duktape/issues/1167#issuecomment-266135612).

ryzhovau, 2018-06-04:
Done via https://github.com/Entware/rtndev/commit/b7721f4f54a65462a40d7f0447dff9c6da150a65

closed 2018-06-04 by ryzhovau