Entware/Entware #81: opkg: Permission denied

morezco, 2018-06-10:
I followed the instructions specified at https://github.com/Entware/Entware-ng/wiki/Install-on-DD-WRT and though most went well I get Permission denied upon trying to run opkg in any way

ryzhovau, 2018-06-11:
Entware-ng is EOL, sorry.
Also, you catch [this issue](https://github.com/Entware/Entware/issues/63) on some DD-WRT builds.

closed 2018-06-11 by ryzhovau

UncleKasra, 2020-05-29:
I have a WL-500G Premium V2 Router and I flash dd-wrt firmware to it. It's version is :
    DD-WRT v3.0-r40559 mega (08/06/19)

I followed the instructions specified at :
    https://github.com/Entware/Entware/wiki/Install-on-DD-WRT-firmware
and though most went well I get Permission denied upon trying to run opkg in telnet.

I run uname -a and result is :
    Linux DD-WRT 2.4.37 #55004 Tue Aug 6 05:02:00 +04 2019 mips DD-WRT

and I run these commands :
[image]

I pluged a 14G USB flash device to my router and I had to format it in one Fat32 partition. because any other format didn't mount to my router. you can see below why :
 [image]

and I tried to solve my possible problem of fat filesystem from there :
    https://askubuntu.com/questions/892172/running-shell-script-from-external-drive-via-live-user/892198#892198
but it did not happen.

sorry about my english that is not very well! :D