Entware/Entware #82: Thanks!

michgeek, 2018-06-12:
Hi guys,

Just here to drop a big thank you to all of you who are maintaining this project. I think people don't say that enough. I have been using those packages for years and it saved me hundred of hours! It allowed me to make my home network safer and provide it with plenty of cool services.

So thank you for all your hard work!

🥇

darkrain88, 2018-09-14:
i use this issue to thank for your hard working,

the packages have been updated to the 9/11/2018/

tianpi, 2020-10-19:
I also want to thank you so much for your work which I find incredibly useful! In addition I recommend putting a "Donate" button (see Paypal) on the homepage/github..

protogen, 2021-04-17:
I would also like to say thank you.  I find this software repository extremely useful.  Your work is very much appreciated.
I second the "Donate" button recommendation by @tianpi.

ryzhovau, 2021-05-04:
Thanks to @RogueScholar for wiki articles! I didn't found your mail so I put few words of gratitude for Peter here.

RogueScholar, 2021-05-04:
@ryzhovau That's really kind of you, I'm just happy that you're pleased with the changes. I've been an Entware user for...so long I can't even remember what year I became one, and I simply did not want to let any more time pass before contributing to the project in some way, so documentation was where I elected to start. I've already asked most of the really dumb questions at this point, so I'm just trying to put into writing the many things you and others in the community have helped me learn along the way. I'm in the Telegram group as well, so if there's anything you've been wanting to add to the Wiki but never got the chance, please let me know and maybe I can make it happen.

Thanks again for the kind words and the many years of hard work keeping this project alive and vibrant for all of us. Take care. 👏🏻

burnbabyburn, 2022-01-18:
Even in 2020 we still benefit from this projects work. Keep up the good work. Thank you!

wellloaded, 2022-05-06:
Amazing work. Please keep it up!