Entware/Entware #83: Requesting PHP Module: MySQL

nobutyeahbutno, 2018-06-19:
Package Requested: php-mysql or php-mysqli. I was originally on entware-ng had php mysql module installed all working well, i then did opkg update and opkg upgrade and i was updated to the new entware but it is missing the php mysql module and I am unable to access phpmyadmin or postfixadmin on my site :( I can attach screenshots of main pages for phpmyadmin and postfixadmin showing mysql module missing for php if requested.

ryzhovau, 2018-06-19:
```
opkg list *mysql*
```

nobutyeahbutno, 2018-07-05:
Thanks that worked. Had to remove php and modules then install mysql mod first followed by other php7 installs :) all working great now. Thanks once again.

closed 2018-07-06 by ryzhovau