Entware/Entware #84: Downgrade rdiff-backup to 1.2.8

gabooh, 2018-06-22:
The rdiff-backup package version is 1.3.3 currently

There are currently two rdiff-backup official versions, both released in March 2009:
- 1.2.8 (stable)
- 1.3.3 (dev/unstable)

While most of the major distributions ship version 1.2.8, entware is shipping 1.3.3, which has no major improvements.
After trying to sync files from a machine running 1.2.8 (ubuntu 14.04 or 16.04 or 18.04, for example) to a machine running 1.3.3 ,I encountered errors due to the versions being incompatible and the transfer was canceled.

Since most distros have version 1.2.8, it would be way more useful for users to have version 1.2.8 available.

ryzhovau, 2018-12-05:
`rdiff-backup` was upgraded (so called "this mess":) by users request.
We not going to fix that package as abandoned for 9 years. I think it's better to remove it at all.

gabooh, 2018-12-12:
I've just read your answer again @ryzhovau and just want to be clear that "this mess" wasn't refering to your packaging work at all. It's more the fact that this tool is unmaintained and I'm still using it anyway.

Following your answer I think I can close this issue.
Thank you!

closed 2018-12-12 by gabooh

tom-ch1, 2019-01-20:
I ran into this one as well, and it is really nasty. For all those who like a solution, I compiled version 1.2.8 for aarch64. It took me a while to understand the Entware build system, but it finally worked. I also renamed the package to rdiff-backup_stable because some versions of opkg won't let you choose which version of a package to install.
[rdiff-backup_stable_1.2.8-1_aarch64-3.10.ipk.zip](https://github.com/Entware/Entware/files/2777137/rdiff-backup_stable_1.2.8-1_aarch64-3.10.ipk.zip)