Entware/Entware #85: Package request: MTprotoproxy

Kirya2k, 2018-06-25:
Package request
Python proxy for Telegram messenger.
https://github.com/alexbers/mtprotoproxy

Littlle MTproxy with native IPV6 and support of advertise a channel. 
So, please, add the package with capabilities for mipselsf-k3.4 platform, thanks a lot.
---
{Describe killer feature of package}

- [X] No similar package already added and/or this one got some new killer feature(s),
- [X] Package is in active development (at least, not abandoned for 5 years or more),
- [X] I successfully tested added package on my {platform}.

---
Please, fill `{}` fields when asking for new package or delete this template in any other case.

ryzhovau, 2018-09-13:
Can't get working `python3-cryptography`:
```
Python 3.7.0 (default, Sep 08 2018, 20:33:38)
[GCC 7.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import cryptography
>>> cryptography.hazmat.primitives.ciphers
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'cryptography' has no attribute 'hazmat'
```
Closing for now.

closed 2018-09-13 by ryzhovau

alexbers, 2018-09-19:
The proper way to use python3-cryptography module is
`from cryptography.hazmat.primitives.ciphers import Cipher, algorithms, modes`

(I am the author of mtprotoproxy)

ryzhovau, 2018-09-20:
Thanks for feedback, already fixed. This requiest came from [here](https://forum.keenetic.net/topic/4876-mtprotoproxy-%D0%B4%D0%BB%D1%8F-entware/) and `mtprotoproxy` is proven as working.