Entware/Entware #86: Package request: python3-urllib3

blackwind, 2018-06-26:
Currently, the `requests` package for Python 3 is useless because there's no `urllib3` package for Python 3:

```
$ python3 -c "import requests"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/opt/lib/python3.6/site-packages/requests/__init__.py", line 43, in <module>
    import urllib3
ModuleNotFoundError: No module named 'urllib3'

$ opkg list | grep urllib
python-requests - 2.18.4-1 - Requests allow you to send HTTP/1.1 requests. You can add headers, form data, multipart files, and parameters with simple Python dictionaries, and access the response data in the same way. It's powered by httplib and urllib3, but it does all the hard work and crazy hacks for you.
python-urllib3 - 1.22-1 - HTTP library with thread-safe connection pooling, file post, and more.
python3-requests - 2.18.4-1 - Requests allow you to send HTTP/1.1 requests. You can add headers, form data, multipart files, and parameters with simple Python dictionaries, and access the response data in the same way. It's powered by httplib and urllib3, but it does all the hard work and crazy hacks for you.
```

Please add `python3-urllib3`.

zyxmon, 2018-06-26:
We will try to add package with next Entware build. Currently you can dry do add module yourself - https://github.com/Entware/Entware/wiki/Self-installation-of-python-modules

zyxmon, 2018-09-16:
@blackwind - have you tried to build it yourself using pip3? (Nativly)

blackwind, 2018-09-16:
Mixing packages from different package managers never ends well. I'm aware, of course, that it can be done, but it's not a real solution to the oversight I highlighted.

zyxmon, 2018-09-16:
I see no reason to add package to the repo that can be installed natively in 5 minutes. I live the issue open and may come back to it in a month or more. May be someone else can make a package.
A lot of python packages were made by users natively ot of the box.

blackwind, 2018-09-16:
By that logic, all Python packages should be removed since they can all be installed natively.

zyxmon, 2018-09-16:
Some python packages can be easily build natively and require a lot of effort for crosscompiling - like building host variant first and target variant at the last step.
We add python packages from trunk (openwrt) and packages that are needed as a dependency for Entware packages, that are not included in openwrt. We have provided a simple instruction to build python packages natively for other cases.

zyxmon, 2018-09-16:
@blackwind - please ask openwrt developers to add python3 variant of this package - https://github.com/openwrt/packages/blob/master/lang/python/python-urllib3/Makefile and it will be added to Entware.

closed 2018-12-27 by ryzhovau