Entware/Entware #87: Go checksum fails, all mirrors are 404s

RNCTX, 2018-06-26:
```
make tools/install -j1 V=s

...

SHELL= flock /home/ddwrt/Entware/tmp/.go1.4.3.linux-386.tar.gz.flock -c '  	/home/ddwrt/Entware/scripts/download.pl "/home/ddwrt/Entware/dl" "go1.4.3.linux-386.tar.gz" "ce3140662f45356eb78bc16a88fc7cfb29fb00e18d7c632608245b789b2086d2" "" "https://storage.googleapis.com/golang"    '
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://storage.googleapis.com/golang/go1.4.3.linux-386.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 47.7M  100 47.7M    0     0  1062k      0  0:00:46  0:00:46 --:--:-- 1130k
Hash of the downloaded file does not match (file: 20c666025dd84dd7f2ec856aa5c0c104b6477c13b564113555b048a65e105306, requested: ce3140662f45356eb78bc16a88fc7cfb29fb00e18d7c632608245b789b2086d2) - deleting download.
```

zyxmon, 2018-06-26:
Jus remove line 18 - https://github.com/Entware/Entware/blob/master/tools/go-bootstrap/Makefile#L18
You are compiling on a 32-bit linux system. Please switch to x64. We have not tested building Entware on 86 distribution for several years.
I am keeping this issue open because Makefile has a bug for i86.

RNCTX, 2018-06-26:
will do, thanks!

last time I messed with arm cross compiling, the qemu tools didn't work with 64 bit hosts, so if they've improved in the years since I'll go on up to 64 bit.

zyxmon, 2018-06-27:
Wу do not use quemu for cross compilation.

closed 2018-09-10 by ryzhovau