Entware/Entware #90: opkg update failed

rozhuk-im, 2018-06-30:
On Samsung SM-G531F Galaxy Grand Prime with http://bin.entware.net/armv7sf-k3.2/
got: opkg_download: Failed to download http://bin.entware.net/.armv7sf-k3.2/Packages.gz, wget returned 255.

 wget works fine, it from busybox.
Also I try to install wget via files (download packages and make repo file:/....) install ok, wget work, but opkg update fail.

No network issue: I try to download Packages.gz and it OK.

No network packets send on opkg update.

Also I try:
http://pkg.entware.net/binaries/armv7/experimental/opkg
http://pkg.entware.net/binaries/armv7/installer/opkg
(on another my device this works OK)

https://github.com/shr-project/opkg/blob/master/libopkg/xsystem.c#L47
I suspect error on execvp() and if replace it to:
`_exit(execvp(argv[0], (char*const*)argv));`
the return code will be more informative - not 255.

Or build with curl should fix it.

zyxmon, 2018-06-30:
Installing on Android is always tricky. Sometime you need to add missing files from linux in Android. resolve.conf is always needed. I would also check length of file your wget downloads from repo.

rozhuk-im, 2018-06-30:
I use modified script: http://www.netlab.linkpc.net/download/tmp/entware_install.sh
```
echo "nameserver 8.8.8.8" > /opt/etc/resolv.conf
...
ln -s /opt/etc/resolv.conf /system/etc/resolv.conf
```
wget from /system/xbin/busybox  work even without this.
wget size OK.

zyxmon, 2018-06-30:
wget from busybox 2.26.2 downloads corrupted files from nginx powered webservers. Please check that it is not your case.

rozhuk-im, 2018-06-30:
Mine busybox is 1.28.3 from f-droid.org repo.
sha256 for all files from http://bin.entware.net/armv7sf-k3.2/installer/ are equal.

rozhuk-im, 2018-07-04:
I found how to fix and why this happen.

https://stackoverflow.com/questions/36674727/permission-denied-on-exec-from-a-binary-running-as-root
Samsung deny exec() from /data, an I place entware to /data/opt, and make simlink: /opt->/data/opt.

Fix: replace simlink to mount:
mount -o rw,bind /data/opt /opt

Solution from:
https://github.com/cSploit/android/issues/1
https://android.stackexchange.com/questions/91650/execve-permission-denied-under-data

Please:
1. https://github.com/shr-project/opkg/blob/master/libopkg/xsystem.c#L47
return (-1); -> return (errno); - to get proper error code in error message.

2. In WiKi add note that:

- Build:
-- minimum free space to build more than 32Gb
-- do not build from root (at least tar build will fail)

- Install
-- on Samsung (at least some devices) entware will not work /data partition because exec() syscall is denied by kernel or workaround required: mount -o rw,bind /data/opt /opt

zyxmon, 2018-07-05:
@rozhuk-im - please add your note to the wiki yourself.

closed 2018-09-10 by ryzhovau