Entware/Entware #91: The monit and monit-nossl packages do not run after installation

fbigun, 2018-06-30:
Error after running the command
```
monit --help
```

Error message
```
AssertException: getpwuid_r failed -- No such file or directory
 raised in init_env at src/env.c:105
```

zyxmon, 2018-07-02:
Not confirmed:
```
 # opkg install monit
Installing monit (5.24.0-1) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/monit_5.24.0-1_armv7-3.2.ipk
Configuring monit.
/share/CACHEDEV1_DATA/.qpkg/Entware/root # monit --help
Usage: monit [options]+ [command]
Options are as follows:
 -c file       Use this control file
....
# uname -a
Linux QNAPD1 3.10.20-al-2.5.3 #66 SMP PREEMPT Tue May 1 04:23:27 CST 2018 armv7l unknown

```

closed 2018-07-02 by zyxmon

droberg, 2019-02-12:
Sorry for waking this up.

I ran into the same issue after upgrading. It seem like a missing /opt/etc/passwd file which throws "whoami" commands off.

See for example:

> strace whoami 2>&1 | grep -E '/etc|/lib'
access("/opt/etc/ld.so.preload", R_OK)  = -1 ENOENT (No such file or directory)
open("/opt/lib/tls/v7l/libgcc_s.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/opt/lib/tls/v7l", 0x7eb945e8)  = -1 ENOENT (No such file or directory)
open("/opt/lib/tls/libgcc_s.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/opt/lib/tls", 0x7eb945e8)      = -1 ENOENT (No such file or directory)
open("/opt/lib/v7l/libgcc_s.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/opt/lib/v7l", 0x7eb945e8)      = -1 ENOENT (No such file or directory)
open("/opt/lib/libgcc_s.so.1", O_RDONLY|O_CLOEXEC) = 3
open("/opt/lib/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
open("/opt/usr/lib/locale/locale-archive", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
open("/opt/etc/nsswitch.conf", O_RDONLY|O_CLOEXEC) = 3
open("/opt/lib/libnss_files.so.2", O_RDONLY|O_CLOEXEC) = 3
open("/opt/etc/passwd", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)

zyxmon, 2019-02-12:
Not confirmed:

```
~ # opkg install coreutils-whoami
Installing coreutils-whoami (8.30-1) to root...
Downloading http://bin.entware.net/mipselsf-k3.4/coreutils-whoami_8.30-1_mipsel-3.4.ipk
Installing coreutils (8.30-1) to root...
Downloading http://bin.entware.net/mipselsf-k3.4/coreutils_8.30-1_mipsel-3.4.ipk
Configuring coreutils.
Configuring coreutils-whoami.
~ # whoami
root
~ # opkg install strace
Installing strace (4.25-1) to root...
Downloading http://bin.entware.net/mipselsf-k3.4/strace_4.25-1_mipsel-3.4.ipk
Configuring strace.
~ # strace whoami 2>&1 | grep -E '/etc|/lib'
access("/opt/etc/ld.so.preload", R_OK)  = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/lib/tls/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/opt/lib/tls", 0x7fafd308)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/lib/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/opt/usr/lib/locale/locale-archive", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
openat(AT_FDCWD, "/opt/etc/nsswitch.conf", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/opt/lib/libnss_files.so.2", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/opt/etc/passwd", O_RDONLY|O_CLOEXEC) = 3

```

droberg, 2019-02-12:
To be more clear:

I suffered the same issue as the original poster after upgrade of entware. This update was the  major one that included the merge with 3x i believe.

I noticed 3 problems

1. Monit didnt start, complains about "getpwuid_r"
2. Telldus daemon coulnt find root group
3. Prompt showed "I dont know who i am" or something similar.

I  am Running on AsusWRT Merlin with  2.6 kernel

The problem for 1 and 3  was missing /opt/etc/passwd and  the problem for 2 was missing /opt/etc/groups

Don't know where they are supposed to originate from but they were as I say missing. I wouldnt expect them to be missing for you since your getpwuid was already working.