Entware/Entware #92: Last version seems to break symlinks (and Samba)

TCB13, 2018-07-02:
Hello,

First of all many thanks fro maintaining this project. It is very, very useful. Now, I've been using entware on a Netgear R7000 and everything was working fine until I decided to run `opkg update; opkg upgrade` last week.

Samba3.6 stopped working right on that moment. I couldn't connect to the server, however it was running. Did some digging into the log files and... found this:
``````
smbd main failed to setup system user info
``````

I'm not sure what package got updated and messed this up however, I also found that the ` /opt/etc/passwd` symlink disappeared. It should be there since:  `in standard case /opt/etc/passwd is a symlink to /etc/passwd file.`

I fixed this issue manually by:
``````
ln -s /etc/passwd /opt/etc/passwd
ln -s /etc/group /opt/etc/group
``````

And now I can't still connect to Samba, but it says the user as authenticated correctly and the error above is gone. Any tips? How can I debug this?

Why are the symlink gone? I also tried to install `shadow*` to fix symlink before trying to manually create them and it didn't work.

TCB13, 2018-07-02:
**Update**: Samba is not working after reboot, however... why did symlinks disappear? Is my fix okay?

zyxmon, 2018-07-02:
What was repo url before update and after update. Probably you are in the middle of converting Entware-ng to Entware. Please post more info about your hardware(feed) and opkg upgrade log.
http://bin.entware.net/Readme.txt

TCB13, 2018-07-02:
@zyxmon thanks for the answer. I'm not sure what URL your talking about, is it the one at `/opt/etc/opkg.conf`? If so: ` http://bin.entware.net/armv7sf-k2.6` before and after.

Unfortunately I don't have the terminal log from the upgrade. But I do have another device, same hardware and entware where I didn't run the upgrade yet. Probably running what I had on the other system, since I've installed the both devices on the same day.

zyxmon, 2018-07-02:
Why have you installed Entware for old kernel? For your kernel you can use armv7sf-k3.2 feed.
I cannot tell what went wrong with Entware upgrade in your case without log. We have no similar reports.
You can save your current Entware in another folder and reinstall using appropriate script from - https://github.com/Entware/Entware/wiki/Alternative-install-vs-standard
You can move package configuration files to the new installation.
Please note that `/opt/etc/passwd` is a symlink for standard installation and a regular file for alternative one.

TCB13, 2018-07-04:
Just a note, I decided to update the other device in similar conditions and I didn't get this error. Everything went just fine.

root@Stardust:~# cat /opt/etc/opkg.conf
src/gz entware http://bin.entware.net/armv7sf-k2.6

root@Stardust:~# uname -a
Linux Stardust 4.4.124 #548 SMP Wed Mar 28 09:52:34 CEST 2018 armv7l DD-WRT

root@Stardust:~# opkg update
Downloading http://bin.entware.net/armv7sf-k2.6/Packages.gz
Updated list of available packages in /opt/var/opkg-lists/entware
root@Stardust:~# opkg upgrade
Upgrading transmission-daemon-openssl on root from 2.93-7 to 2.94-1...
Downloading http://bin.entware.net/armv7sf-k2.6/transmission-daemon-openssl_2.94-1_armv7-2.6.ipk
Upgrading transmission-web on root from 2.93-7 to 2.94-1...
Downloading http://bin.entware.net/armv7sf-k2.6/transmission-web_2.94-1_armv7-2.6.ipk
Configuring transmission-daemon-openssl.
Configuring transmission-web.
Collected errors:
 * resolve_conffiles: Existing conffile /opt/etc/transmission/settings.json is different from the conffile in the new package. The new conffile will be placed at /opt/etc/transmission/settings.json-opkg.

So I'm closing this. Thank you very much.

closed 2018-07-04 by TCB13